OpenTabletDriver/OpenTabletDriver #3254: XP-Pen Artist 16 (gen 2) reports hight presure value

naproxes, 2024-03-18:
### Steps to Reproduce

1. Install driver
2. Open debug
3. Put Pen down on Tablet
4. Switch to eraser and put down

### Expected Behavior

I get values from 0-8191 something

### Observed Behavior

I get values ranging from 4294959104 - 4294967295

### Tablet Device

XP-Pen Artist 16 (gen 2)

### Diagnostic Information

[tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14642516/tablet-data.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-08-08:
Is there no firmware update available?

jamesbt365, 2024-11-20:
Should be fixed by #3455 and the fix will be available on the next release, if it isn't, let me know and I'll reopen this issue.

closed 2024-11-20 by jamesbt365