OpenTabletDriver/OpenTabletDriver #3255: Huion H580x not detecting

CheweyBoi08, 2024-03-19:
### What do you need help with?

Hello my Huion H580x is not being detected i know it is quirky but it just isnt being detected

### What have you tried?

I have used both windows and linux. Linux being my main operating system and have re plugged my tablet for almost an hour and nothing is working 
[Dump1.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14657630/Dump1.txt)

### Diagnostic Information

[Diagnostics.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14657639/Diagnostics.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-03-19:
Place [this](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14657663/H580X.json) file in `
~/.local/share/OpenTabletDriver/Configurations`.

Replug your tablet, then go into `tablet > tablet debugger`, i'll need the following measurements:
- The Max X and Max Y obtained by moving your pen to the bottom right of the physical tablet (ignore the visualizer).
- The max pressure when you press the pen down all the way.
- If the pen buttons show true when you press them.
- If the tablet buttons do the same.

CheweyBoi08, 2024-03-20:
max X and Y is 40640, 25400 pressure is 8191 the pen buttons are registering and so does the tablet buttons

PR #3256 (2024-04-12): Add support for variant of Huion H580X
Fixes #3255 

verification: 
[Dump1.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14657653/Dump1.txt)
[Diagnostics.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14657654/Diagnostics.json)
closed 2024-04-12 by InfinityGhost