OpenTabletDriver/OpenTabletDriver #3259: Support Request: How do you perform a right-click?

SimonEast, 2024-03-21:
### What do you need help with?

How do you perform a right-click with OpenTabletDriver? Before I installed OTD, I could right-click in Windows by just tapping and holding for 1 second with the pen. This no longer works when I'm using OpenTabletDriver. I tried several things, but just couldn't figure it out.

Is this functionality even available? There doesn't seem to be any settings or options for it within the app.

I'm using a Gaomon 1060pro.

### What have you tried?

* Tapping and holding
* Pressing various combinations of buttons

### Diagnostic Information

[1060pro.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14680743/1060pro.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

Spencer-Sawyer, 2024-03-21:
Have you tried enabling windows-ink in the driver gui?

SimonEast, 2024-03-21:
> Have you tried enabling windows-ink in the driver gui?

Thanks for the reply. I've checked through all the tabs inside OTD and I don't seem to have that option. I read through the FAQ, and it mentions installing a VMulti Driver and a WindowsInk plugin for OTD - do I need those?

As a workaround, I've discovered that I can bind one of the pen buttons to be the right button, under Pen Settings > Pen Buttons > Pen Binding 1 & 2. That might do for some uses, although it would be good to know if the tap-and-hold method is also possible.

SimonEast, 2024-03-21:
I ended up installing the VMulti driver and WindowsInk plugin, and once I switched to "Windows Ink Absolute Mode" this allowed the ability to tap-and-hold for a right click, so this somewhat solves the problem. However it doesn't seem possible to _also_ have use the pen-button for a right-click under WindowsInk mode, so it seems I can't have both. But at least this will do.

I also had to check some of the settings under
_Windows Control Panel (the old classic one, not the new one) > View by icons > Pen and Touch > Press and hold > Settings_
and make sure the tap-and-hold functionality was correctly enabled.

Hope this helps others with this issue.

gonX, 2024-03-23:
We don't have tap and hold to right click in the driver. But you can bind a pen button to right click.

Please open a feature request instead if you would like tap and hold to right click natively supported.

closed 2024-03-23 by gonX