OpenTabletDriver/OpenTabletDriver #3260: Side buttons mouse bindings not working (Arch Linux)

Javatrix, 2024-03-21:
### Steps to Reproduce

1. Open the driver
2. Go to the pen settings tab
3. Set any of the pen buttons (not tip) to a mouse button (at least with Huion HS611)

### Expected Behavior

Mouse clicks should register just fine.

### Observed Behavior

The clicks are not sent, not only to Krita, but any other program as well.

### Tablet Device

Huion HS611

### Diagnostic Information

[diagnostic..json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14704307/diagnostic.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-03-21:
Restriction of pens on linux.

Artist mode emulates a tablet directly, absolute mode emulates an absolute mouse, thus can use mouse buttons.

closed 2024-03-21 by jamesbt365

DmitryFX, 2024-08-21:
I have the same problem on win 10. 

Huion H610X.
Windows Ink Relative Mode.
Cannot bind mouse buttons to pen buttons.
If I bind any letter to pen button, it works as expected, prints letters in the text editor. So, pen button detection works.

Any way to fix this?

jamesbt365, 2024-08-21:
You can fix that by opening the plugin manager, navigating to the vmulti plugin and pressing "show plugin wiki", before reading it. It is important that you read these entries as they include setup instructions.

Javatrix, 2024-08-21:
Haven't found any, I had to rebind the shortcut to a key.

DmitryFX, 2024-08-21:
> You can fix that by opening the plugin manager, navigating to the plugin and pressing "show plugin wiki", before reading it. It is important that you read these entries as they include setup instructions.

In VMulti mode mouse bindings work, but Windows Ink Pen Tip stopped working.
Finally, I can have either pressure control or mouse button bindings, not together?

jamesbt365, 2024-08-21:
regular mouse buttons for regular mouse modes
artist mode bindings for artist mode
vmulti bindings for vmulti modes
windows ink bindings for windows ink modes

DmitryFX, 2024-08-21:
Thank you for the answers!

Suddenly, it started functioning as expected (pressure sensitivity + mouse buttons bindings).
Very strange, I believe, I've already tried this combination:

Mode: Windows Ink Relative Mode.
Tip binding: Windows Ink - pen tip.
Pen binding 1: VMulti Mode (Button: Right)