OpenTabletDriver/OpenTabletDriver #3265: Request for Permission to Publish Opentabletdriver as a Flatpak on Flathub

first-storm, 2024-03-25:
### Description

I am reaching out to request permission to package and publish Opentabletdriver as a Flatpak on Flathub. This would significantly ease the installation and use of Opentabletdriver for Linux users. Packaging the software as a Flatpak is particularly advantageous for dotnet programs, as it allows users to operate the software with just the installation of udev configurations on the host machine, eliminating the need to go through the cumbersome process of installing dotnet. Additionally, it provides convenience for users of musl libc.

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

InfinityGhost, 2024-03-26:
Relevant PR: https://github.com/flathub/flathub/pull/5082

jamesbt365, 2024-03-26:
Got word from the main developers of OpenTabletDriver and we are willing to give the green-light on this if @X9VoiD and @InfinityGhost can get write access on the repository on Flathub because its such an influential packaging format.

Thank you for creating this package so far, I haven't looked into how it ties into things too much or how flatpak really works but I we understand this is an important package to provide, and this would fix #2708.

first-storm, 2024-03-26:
> Got word from the main developers of OpenTabletDriver and we are willing to give the green-light on this if the main developers (@X9VoiD and @InfinityGhost) can get write access on the repository on Flathub because its such an influential packaging format.
> 
> Thank you for creating this package so far, I haven't looked into how it ties into things too much or how flatpak really works but I we understand this is an important package to provide, and this would fix #2708.

The main developers of OpenTabletDriver naturally have read and write access to this repository. It's a Flathub policy.

first-storm, 2024-03-28:
It appears the Flathub Reviewer forgot to grant the upstream developers access permissions. Please consider opening an issue on the Flathub repository. Guidance can be found here: https://docs.flathub.org/docs/for-app-authors/maintenance/#maintainer-access-to-flathub-application-repository.

closed 2024-05-13 by InfinityGhost