OpenTabletDriver/OpenTabletDriver #3268: Add support for XP-Pen Artist 24 Pro

MichaelDevon, 2024-03-30:
### Manufacturer

XP-Pen

### Product Name

Artist 24 Pro

### Device Width

632mm

### Device Height

370mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Diagnostic Information

```json {
  "App Version": "OpenTabletDriver v0.6.4.0+9f4e357910a5917ef86166c195c858c1c65b46d2",
  "Build Date": "12/22/2023",
  "Operating System": {
    "Name": "Ubuntu",
    "Version": "23.10 (Mantic Minotaur)",
    "Attributes": {
      "PRETTY_NAME": "Ubuntu 23.10",
      "VERSION_ID": "23.10",
      "VERSION_CODENAME": "mantic",
      "ID": "ubuntu",
      "ID_LIKE": "debian",
      "KERNEL_VERSION": "6.5.0-26-generic"
    }
  },
  "Environment Variables": {
    "USER": "user",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": null,
    "PWD": "/home/user",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.3/0000:01:00.0/usb1/1-5/1-5.3/1-5.3:1.0/0003:28BD:092D.0003/hidraw/hidraw2",
      "Manufacturer": "UGTABLET",
      "ProductName": "24 inch PenDisplay",
      "SerialNumber": "",
      "FriendlyName": "24 inch PenDisplay",
      "VendorID": 10429,
      "ProductID": 2349,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORTS": "09:0001:0002, 01:0001:0002, 06:0001:0006"
      }
    },
    {
[... 2126 lines cut ...]
      "Notification": false
    },
    {
      "Time": "2024-03-30T22:38:17.2580234+01:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-03-30T22:38:20.4096206+01:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
} ```

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Aroy-Art, 2024-06-10:
I would like to se support for the tablet, I own one so if there is anything i can do to help I'm happy to.

I did see there is some work going on for it 827a4ce

Aroy-Art, 2024-06-10:
I have got it to "work" with the [aur git package (v0.7.0.0pre.r893.g3dfaddf3)](https://aur.archlinux.org/packages/opentabletdriver-git)

tho i cant remap any buttons 

Here is the log
```
[RootHub:Debug]	Initialized internal child hubs: HidSharpDeviceRootHub
[Detect:Info]	Searching for tablets...
[Detect:Debug]	Attempting to match config 'XP-Pen Artist 24 Pro' to device '24 inch PenDisplay (/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-3/3-3.3/3-3.3:1.2/0003:28BD:092D.0018/hidraw/hidraw7)'
[Detect:Debug]	Found 'XP-Pen Artist 24 Pro' digitizer: '24 inch PenDisplay (/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-3/3-3.3/3-3.3:1.2/0003:28BD:092D.0018/hidraw/hidraw7)'
[Detect:Debug]	Attempting to match config 'XP-Pen Artist 24 Pro' to device '24 inch PenDisplay (/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-3/3-3.3/3-3.3:1.1/0003:28BD:092D.0017/hidraw/hidraw6)'
[Detect:Debug]	Attempting to match config 'XP-Pen Artist 24 Pro' to device '24 inch PenDisplay (/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-3/3-3.3/3-3.3:1.0/0003:28BD:092D.0016/hidraw/hidraw5)'
[Detect:Info]	Found tablet 'XP-Pen Artist 24 Pro'
[Device:Debug]	Initializing device '24 inch PenDisplay (/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:08:00.3/usb3/3-3/3-3.3/3-3.3:1.2/0003:28BD:092D.0018/hidraw/hidraw7)'
[Device:Debug]	Using report parser type 'OpenTabletDriver.Configurations.Parsers.XP_Pen.XP_PenReportParser'
[Device:Debug]	Set device output: 02-B0-04
[Evdev:Debug]	Successfully initialized virtual pressure sensitive tablet. (code NONE)
[LinuxArtistMode:Debug]	Input: [526,85x296,35@<263,425  148,175>:0°]
[LinuxArtistMode:Debug]	Output: [2560x1440@<4880  750>]
[LinuxArtistMode:Debug]	LockAspectRatio: False
[LinuxArtistMode:Debug]	AreaClipping: True
[LinuxArtistMode:Debug]	AreaLimiting: True
[LinuxArtistMode:Debug]	LockToBounds: True
[XP-Pen Artist 24 Pro:Info]	Output mode: Artist Mode
[MouseBinding:Debug]	Button: Left
[MouseBinding:Debug]	Button: Middle
[MouseBinding:Debug]	Button: Right
[Settings:Info]	Driver is enabled.
[IPC:Debug]	Connected to a client.

```

closed 2024-06-10 by jamesbt365

Aroy-Art, 2024-06-11:
> > tho i cant remap any buttons
> 
> User error, you have not saved appropriate settings.

I got it to work now, had a broken udev rule from testing DIGImend

But the tablet has 20 buttons and i can only map 8 why is that?

jamesbt365, 2024-06-11:
#3335.

Aroy-Art, 2024-06-11:
> #3335.

Amazing :purple_heart:, keep up the amazing work