OpenTabletDriver/OpenTabletDriver #3270: Add support for Wacom CTL 6100WL to Osu! Lazer

NoxianBrainBird, 2024-04-02:
### Manufacturer

Wacom

### Product Name

CTL-6100WL

### Device Width

264mm x 200mm   W x L

### Device Height

8.8mm H

### Pressure Levels

4096

### Lines per Inch (LPI)

2540

### Additional Information

I'm not 100% sure about the diagnostic info, i used OpebTabletDriver and followed the instructions but it had lots of information on it and I wasn't sure what i needed so i simply pasted the file

### Diagnostic Information

[Waccom CTL 6100WL .txt.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14831225/Waccom.CTL.6100WL.txt.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-04-02:
Your tablet is [supported already](https://opentabletdriver.net/Tablets). The reason its not showing up is because you have the Wacom Driver installed which blocks all instances of OTD from working, if you want to use the inbuilt OTD or just OTD overall, you must uninstall the Wacom Driver, which unless you replace with  OTD systemwide will leave you no tablet driver, pick your poison.

closed 2024-04-02 by jamesbt365

NoxianBrainBird, 2024-04-02:
> <https://github.com/OpenTabletDriver/OpenTabletDriver/issues/3270> as not
> <OpenTabletDriver/OpenTabletDriver/issue/3270/issue_event/12322978457@
> github.com>

Oh okay, Thankyou for the quick response and apologies for maybe the stupid
question

Regards
NoxianBrainBird