OpenTabletDriver/OpenTabletDriver #3274: Support Request:

YoruNoKen, 2024-04-07:
### What do you need help with?

I installed OpenTabletDriver via the AUR on Arch Linux, but it fails to detect my tablet.
There are no errors in the console.

### What have you tried?

I tried restarting my computer, building OpenTabletDriver from source, restarting daemon

### Diagnostic Information

```json
[
{
  "Time": "2024-04-07T21:13:37.8093144+03:00",
  "Group": "Detect",
  "Message": "Searching for tablets...",
  "StackTrace": null,
  "Level": 1,
  "Notification": false
},
{
  "Time": "2024-04-07T21:13:37.8230243+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet '10moon 1060N'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2024-04-07T21:13:37.8241198+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Acepen AP 1060'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2024-04-07T21:13:37.824141+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Adesso Cybertablet K8'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2024-04-07T21:13:37.8241475+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Artisul A1201'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2024-04-07T21:13:37.8241573+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Artisul AP604'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2024-04-07T21:13:37.8241621+03:00",
  "Group": "Detect",
  "Message": "Searching for tablet 'Artisul M0610 Pro'",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
[... 1951 lines cut ...]
{
  "Time": "2024-04-07T21:13:51.9551471+03:00",
  "Group": "IPC",
  "Message": "Connected to a client.",
  "StackTrace": null,
  "Level": 0,
  "Notification": false
},
{
  "Time": "2024-04-07T21:13:52.5257349+03:00",
  "Group": "Settings",
  "Message": "Presets have been refreshed. Loaded 0 presets.",
  "StackTrace": null,
  "Level": 1,
  "Notification": false
},
```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-04-07:
"My tablet" is functionally not enough information if you fail to provide a diagnostic, I have no idea *what* device is plugged in at all, theres no information for me to go on other than *it isn't* working.

State your tablet and export then upload the diagnostic file when you do help > export diagnostics in the UI.

YoruNoKen, 2024-04-07:
Sorry, I've got a Wacom CTL 672.
I've read over the FAQ. 
The logs used to throw me the insufficient perms error, and after I uninstalled, rebooted, reinstalled from source using mkpkg, and reboot again, it doesn't throw any errors but doesn't detect my tablet anyways.
I've tried using the search function on  GitHub issues and I'm pretty sure I haven't missed it.
I guess the title somehow disappeared because I wrote it fully, sorry about that!

the diagnostics JSON file is over here: https://yorunoken.s-ul.eu/YrHCVK5B

jamesbt365, 2024-04-07:
Your tablet isn't plugged in and recognized as a HID device, theres nothing for OTD to open.

closed 2024-04-09 by jamesbt365