OpenTabletDriver/OpenTabletDriver #3275: VK640 Doesn't after waking from sleep

visayanaryan, 2024-04-09:
### Steps to Reproduce

1. Ensure that the tablet works
2. Put the laptop to sleep
3. Wake up the laptop and test the tablet

### Expected Behavior

Tablet input works after waking laptop from sleep

### Observed Behavior

Tablet doesn't emit any input (pen, buttons). Tablet is detected by the driver but the tablet debugger shows nothing:

[image]

I can see that sleep is detected by OTD but only clicking "Detect tablet" solves the issue

### Tablet Device

VEIKK VK640

### Diagnostic Information

```json
{
  "AppVersion": "OpenTabletDriver v0.7.0.0-pre",
  "BuildDate": "",
  "OperatingSystem": {
    "Name": "Arch Linux",
    "Version": "Unknown",
    "Attributes": {
      "PRETTY_NAME": "Arch Linux",
      "ID": "arch",
      "BUILD_ID": "rolling",
      "KERNEL_VERSION": "6.8.4-arch1-1"
    }
  },
  "EnvironmentVariables": {
    "USER": "user",
    "DISPLAY": ":0",
    "PWD": "/home/user",
    "PATH": "/usr/local/sbin:/usr/local/bin:/usr/bin"
  },
  "Devices": [
    {
      "VendorID": 12267,
      "ProductID": 5,
      "InputReportLength": 13,
      "OutputReportLength": 9,
      "FeatureReportLength": 0,
      "Manufacturer": "VEIKK.INC",
      "ProductName": "VK640",
      "FriendlyName": "VK640",
      "SerialNumber": "0000001",
      "DevicePath": "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7.4/3-7.4:1.2/0003:2FEB:0005.0007/hidraw/hidraw6",
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "2",
        "HID_REPORTS": "09:FF0A:0001"
      }
    },

    {
      "VendorID": 12267,
      "ProductID": 5,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "Manufacturer": "VEIKK.INC",
[... 480 lines cut ...]
      "Message": "Tablet debugging is enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-04-09T05:12:04.2204675+00:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is disabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    }
  ]
}

```

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

closed 2024-05-10 by jamesbt365

gonX, 2024-10-19:
I don't think this is a duplicate issue. #3171 does not detect the sleep, and is on v0.6.4.0, which does not have the improved sleep detection that `master` has, which this issue uses, and this issue has the sleep detected.

@visayanaryan if you can reproduce this on latest `master` please reopen the issue.

visayanaryan, 2025-01-31:
Ugh I completely missed this one. Yep, last time I updated (few days ago) I faced the same bug. Re-detect or re-plug solves the issue. I will try latest commit in few minutes just to be sure. I can't re-open this btw (no permission)

gonX, 2025-12-05:
`master` is now EOL. If you can reproduce this on `0.6.x` you may reopen this issue.