OpenTabletDriver/OpenTabletDriver #3279: Adding OpenTabletDriver to the scoop extras bucket

make-42, 2024-04-12:
### Description

Hello. Have you ever considered adding OpenTabletDriver to the scoop extras bucket to facilitate installation for users? 
[This](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14963992/OpenTabletDriver.json) is a working package JSON manifest for OpenTabletDriver you can use or build upon if you ever decide to add OpenTabletDriver.

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-04-12:
Did you even check if there was a package? Its not officially maintained and we don't have intentions to officially maintain it at this time but a package *does* exist in the extras bucket.

https://github.com/ScoopInstaller/Extras/blob/master/bucket/opentabletdriver.json

make-42, 2024-04-12:
i did but could not find it. sorry.