OpenTabletDriver/OpenTabletDriver #3280: Veikk A15 has no aux button support

adhdandy, 2024-04-13:
### Description

I have made the necessary changes to fix the issue for myself, but I don't know how to implement them in a way that would work as a PR for the project without breaking support for other Veikk tablets.
Could someone look at the changes I made in my fork and add them properly?

Fork: https://github.com/adhdandy/OpenTabletDriver

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-04-13:
Just submit a parser under a new name and evidence that it works, provide data recordings of movement on both the X and Y axis in your PR alongside pressure if possible.

jamesbt365, 2024-08-09:
#3284.

PR #3284 (2024-08-03): Fix aux button support on Veikk A15 tablets
Fixes #3280.

Proof [tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15000732/tablet-data.txt)
closed 2024-08-09 by jamesbt365