OpenTabletDriver/OpenTabletDriver #3281: Add support for XP-Pen Deco Pro LW gen 2

AT-Geist, 2024-04-14:
### Manufacturer

XP-Pen

### Product Name

MT1172B

### Device Width

333mm

### Device Height

258.1mm

### Pressure Levels

16384

### Lines per Inch (LPI)

5080

### Diagnostic Information

[DecoProLWG2.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14969138/DecoProLWG2.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

AT-Geist, 2024-04-15:
Added in #3228

closed 2024-04-15 by AT-Geist

DENG-XC, 2024-12-27:
Hello, I'm thinking about to buy this tablet. I saw on Reddit that the official drivers don't support Bluetooth on Linux. Just wondering, does OTD have Bluetooth support for this tablet?

jamesbt365, 2024-12-27:
It doesn't.