OpenTabletDriver/OpenTabletDriver #3282: Error with PluginManager

Ostenarchy, 2024-04-16:
### Steps to Reproduce

When i open PluginManager it give me this error.
1) I use the latest version of OpenTabletDriver
2) Before installation I used the "TabletDriverClean"
3) I install this version of .NET: [https://opentabletdriver.net/Framework](url) (Windows x64)
4) My OS is Windows LTSC (screenshot down below)

[image]
[image]
[image]

### Expected Behavior

I expect of plugin manager will open.

### Observed Behavior

It gives me the error message and do not open.

### Tablet Device

Wacom CTL-480

### Diagnostic Information

[DIAGNOSTIC.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/14988092/DIAGNOSTIC.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-04-16:
Your computer was unable to make a connection to GitHub correctly.

jamesbt365, 2024-04-16:
If you can't get it working, youll need to manually install plugins from our website.

closed 2024-04-16 by jamesbt365