OpenTabletDriver/OpenTabletDriver #3283: Support Request: no tablets are detected | fedora linux 39 | HUION H640P

sudo-kill, 2024-04-16:
### What do you need help with?

the software does not see the tablet,

[Detect:Info]	No tablets were detected.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.

### What have you tried?

reinstalled the software
rebooted the computer
deleted and downloaded the official drivers for the tablet

i've tried a lot of things, i can't solve this problem for 3 days

### Diagnostic Information

[Detect:Info]	No tablets were detected.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

sudo-kill, 2024-04-16:
`[InvalidOperationException:Error]	Collection was modified; enumeration operation may not execute.
   at System.Collections.Generic.List`1.Enumerator.MoveNext()
   at System.Linq.Enumerable.WhereListIterator`1.MoveNext()
   at OpenTabletDriver.Driver.MatchDevice(TabletConfiguration config, IList`1 identifiers) in /__w/OpenTabletDriver/OpenTabletDriver/dist/redhat/opentabletdriver/BUILD/opentabletdriver-0.6.4.0/OpenTabletDriver/Driver.cs:line 149
   at OpenTabletDriver.Driver.Match(TabletConfiguration config) in /__w/OpenTabletDriver/OpenTabletDriver/dist/redhat/opentabletdriver/BUILD/opentabletdriver-0.6.4.0/OpenTabletDriver/Driver.cs:line 138
`

jamesbt365, 2024-04-16:
The information you have provided is not enough to diagnose the issue, I need tge file generated from `help > export diagnostics`

sudo-kill, 2024-04-16:
> The information you have provided is not enough to diagnose the issue, I need tge file generated from `help > export diagnostics`

[logs.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15000305/logs.json)

jamesbt365, 2024-04-16:
Your tablet isn't plugged in/detectable as valid HID devices. OTD can't detect what it can't see.

Ensure that:
- The tablet is plugged in.
- The cable its using supports data transfer and not just power delivery.
- The USB port and or connector is not damaged.

closed 2024-04-16 by jamesbt365

sudo-kill, 2024-04-16:
> Your tablet isn't plugged in/detectable as valid HID devices. OTD can't detect what it can't see.
> 
> Ensure that:
> 
> * The tablet is plugged in.
> * The cable its using supports data transfer and not just power delivery.
> * The USB port and or connector is not damaged.

thank you for answer, the cable is connected to the network, I can use a tablet with drivers from the manufacturer, I can also see the tablet using the lsusb command

what can I do in this case?

jamesbt365, 2024-04-16:
Export another diagnostic, if OTD can still not see your device then it probably isn't setting up properly.

Phinguus, 2024-04-16:
This is also happening to me, tablet stopped being detected randomly today after using OpenTabletDriver without issues for months, I'm using the Gaomon S620 and I've tried all in the FAQs
[log.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15003732/log.json)

jamesbt365, 2024-04-16:
> This is also happening to me, tablet stopped being detected randomly today after using OpenTabletDriver without issues for months, I'm using the Gaomon S620 and I've tried all in the FAQs [log.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15003732/log.json)

Unrelated issue. https://opentabletdriver.net/Wiki/FAQ/Windows#driver-cleanup

sudo-kill, 2024-04-17:
> Unrelated issue. https://opentabletdriver.net/Wiki/FAQ/Windows#driver-cleanup

I have Linux, it didn't help me

sudo-kill, 2024-04-17:
> Unrelated issue. https://opentabletdriver.net/Wiki/FAQ/Windows#driver-cleanup
how does the tablet search work?

maybe if I rename the usb it will be able to find it?
[image]

jamesbt365, 2024-04-17:
> I have Linux, it didn't help me

Which is why I quote replied to make it clear and said *unrelated issue*
 
> maybe if I rename the usb it will be able to find it?

That wouldnt work and also you can't just do that, OTD needs valid HID devices, there's nothing for OTD to actually detect and use otherwise.

sudo-kill, 2024-04-18:
> Which is why I quote replied to make it clear and said _unrelated issue_
>
>
> That wouldnt work and also you can't just do that, OTD needs valid HID devices, there's nothing for OTD to actually detect [...]

i get this error now

[InvalidOperationException:Error]	Collection was modified; enumeration operation may not execute.
   at System.Collections.Generic.List`1.Enumerator.MoveNext()
   at System.Linq.Enumerable.WhereListIterator`1.MoveNext()
   at OpenTabletDriver.Driver.MatchDevice(TabletConfiguration config, IList`1 identifiers) in /__w/OpenTabletDriver/OpenTabletDriver/dist/redhat/opentabletdriver/BUILD/opentabletdriver-0.6.4.0/OpenTabletDriver/Driver.cs:line 149
   at OpenTabletDriver.Driver.Match(TabletConfiguration config) in /__w/OpenTabletDriver/OpenTabletDriver/dist/redhat/opentabletdriver/BUILD/opentabletdriver-0.6.4.0/OpenTabletDriver/Driver.cs:line 138

jamesbt365, 2024-04-18:
Looks like the devices changed while OTD was reading them, I don't think that is a hard error.

> I can use a tablet with drivers from the manufacturer

If you have other drivers installed on Linux OTD can't work, thats just how things go. OTD is supposed to be the sole driver on the system.