OpenTabletDriver/OpenTabletDriver #3290: Add support for Huion Inspiroy 2M (H951P)

jouneee, 2024-04-20:
### Manufacturer

Huion

### Product Name

Huion Inspiroy 2M (H951P)

### Device Width

221mm

### Device Height

138mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

This tablet has 8 programmable buttons, a scroller wheel and 3 group buttons

### Diagnostic Information

[diag.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15048285/diag.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-04-20:
Tablet > device string reader
9580
103
dump all, save that file and send it here.

jamesbt365, 2024-04-20:
Looks like this will already be completed with #2927.

closed 2024-07-30 by jamesbt365