OpenTabletDriver/OpenTabletDriver #3297: After installing driver tablet pen only moves on X axis

DiogoSantosTeixeira, 2024-04-24:
### Steps to Reproduce

1. Using Nobara Linux plugged tablet and pen moved correctly right away 
2. Installed OTD following instructions for fedora and the tablet was detected
3. Also followed instructions on how to autostart OTD

### Expected Behavior

Be able to move pen cursor

### Observed Behavior

Pen moves across the X axis but is stuck in the Y axis, so it moves in a single horizontal line across the very top of the screen.

### Tablet Device

XP-Pen G430S

### Diagnostic Information

Exported diagnostics
[diagnostictablet.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15097546/diagnostictablet.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-04-24:
Could you record the debugger?

DiogoSantosTeixeira, 2024-04-24:
I think i did it, is it this ?
[tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15098146/tablet-data.txt)

jamesbt365, 2024-04-24:
Did it work fine before you installed OTD? Because your tablet isn't reporting changes in the y-axis data, its just statically using the same values.

DiogoSantosTeixeira, 2024-04-24:
Yeah it worked fine with whatever controller Nobara has by default, i just plugged it in for the first and it worked. Only thing was that i could not change the active area.

jamesbt365, 2024-04-24:
Well OTD can't really affect the raw data from the tablet, the tablet debugger just shows whats coming in from the tablet.

Unless this is a different variant of the tablet (i highly doubt this, nobody else has reported this, and the identifiers are the same) that require different inits or something I wouldn't be too sure that it is a configuration issue.

Are you sure you don't have anything like a wireless charger, other charging/power bricks around the tablet? Try and isolate as much technology away from the tablet itself.

DiogoSantosTeixeira, 2024-04-24:
Well the tablet is not near anything really but just to be sure i tried unplugging everything else (cause i have a bluetooth dongle and a wireless mouse) with no luck, also tried different usb ports. 
About the variant of the tablet, in OTD says "XP-Pen Star G430S V2" and at the bottom of the tablet says "SKU: StarG430S_B" idk if that checks out .
Also tried OTD in my other drive with Mint installed (not a fresh install) and it behaved exactly the same.

jamesbt365, 2024-04-24:
You're on Linux, WinUSB is a Windows thing. Feel free to Pcap the OEM driver and see what the OEM driver is sending.

DiogoSantosTeixeira, 2024-04-24:
I'm sorry, still new to linux.
Just finished the entire install guide testing the tablet at every step. The tablet began behaving like i described right after i ran "systemctl --user enable opentabletdriver.service --now". Before that it was moving correctly.
Also when run "libwacom-list-local-devices" i get:
devices:
- name: 'XP-Pen G430S'
  bus: 'usb'
  vid: '0x28bd'
  pid: '0x0913'
  nodes: 
  - /dev/input/event2: 'UGTABLET 4 inch PenTablet Mouse'
  - /dev/input/event4: 'UGTABLET 4 inch PenTablet'

jamesbt365, 2024-04-25:
We would need a USB packet capture on windows with the OEM driver to diagnose the actual reasoning behind it not working, as it stands the tablet is not reporting the Y axis correctly, the raw data indicates its locked to a specific value while in range.

Or, you could try messing around yourself with possible initialization processes, there is a chance specifically your tablet doesn't respond properly to the standard init process.

closed 2024-06-04 by DiogoSantosTeixeira