OpenTabletDriver/OpenTabletDriver #3300: I can do testing to add support for the XP-Pen Artist 13.3 Pro's Wheel

Nukley, 2024-04-27:
### Description

I have an Artist 13.3 Pro tablet, I have found this driver to be useful for drawing in the past, however once I got a new tablet, one thing started to bug me... The wheel is not supported! I actually did look in the tablet debugger and found the specific value that is changing.

[image]
This value changes to 1 if rotating clockwise, constantly going between 1 and 0, 0 between each tick, and 1 meaning it got an input. This is very brief and I had to stare at it for a while to see.

Same deal for counterclockwise, except instead of 1 and 0, it's 2 and 0.

Let me know if you need any more information or anything else from me! Thanks!

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-04-27:
OTD doesn't support any wheels at all right now, it simply is missing the infrastructure to handle them. Lets keep things clean and keep everything in one issue. See #1367

gonX, 2026-03-04:
You need to provide a data recording containing only the following 2 movements:

- 2 full clockwise circles
- 2 full counterclockwise circles

(the circles don't need to be perfect)

If they're sliders ("bars") you can do 2 full swipes in both directions instead.

You can record the raw data by opening the OpenTabletDriver interface, and then going to Tablets -> Tablet Debugger. There will then be a "Enable Data Recording" checkbox on the right hand of the new window. Then a `tablet-data`  file will be available in your [app data](https://opentabletdriver.net/Wiki/FAQ/General#appdata) folder.

closed 2026-03-11 by gonX