OpenTabletDriver/OpenTabletDriver #3312: Linux: Artist Mode: No pressure sensitivity

winteriscariot, 2024-05-08:
### Steps to Reproduce

OTD detects my CTL-4100WL when connected via USB (i don't use it wirelessly)

Enabled Artist Mode

Attempted to draw something in MyPaint or Krita

### Expected Behavior

Pressure sensitivity

### Observed Behavior

No pressure sensitivity

### Tablet Device

Wacom CTL-4100WL

### Diagnostic Information

[diag.otd.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15254199/diag.otd.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-05-08:
This shouldn't be a bug within OTD, you should be able to see pressure working correctly in `sudo libinput debug-tablet`. If thats the case everything is reporting correctly on our end and this is a problem with the application you are drawing in.

winteriscariot, 2024-05-08:
My mistake, this is an issue with my compositor, labwc. Pressure works fine in sway.

Related issue in labwc: https://github.com/labwc/labwc/issues/1672

closed 2024-05-08 by winteriscariot