OpenTabletDriver/OpenTabletDriver #3318: Support Request: Install for all users in shared classroom/lab environment?

LRSFC-DanJ, 2024-05-16:
### What do you need help with?

We use a variety of graphics tablets in our Art, Graphic Design and Media classrooms and some of these are no longer supported by the latest official driver. Wacom has also made changes to their driver such that the latest driver and the older generation driver can no longer co-exist. When investigating potential solutions we came across the OpenTabletDriver project. However, the Windows install instructions recommend putting the software in the user's profile area, which is not ideal for computers which can be logged on to by any user, and there also does not appear to be any silent or unattended deployment method, which would cause difficulties when deploying the software to classrooms full of computers.

Is it possible to silently and unattendedly deploy OpenTabletDriver to multiple computers for use by all users on those computers, without needing to put a copy of the software in every user's profile?

### What have you tried?

We are currently still checking whether our tablets are supported by the software.

### Diagnostic Information

This issue is an enquiry not a bug, therefore diagnostic info is not applicable.

### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [ ] I am on the latest version of OpenTabletDriver.

Deftera186, 2024-05-24:
You don't really have to install it in the example folder, but the settings will still be per-user and not system-wide. Also, here's an example script of how to silently install it on the C drive of a system:
```ps
winget install --disable-interactivity -e --id Microsoft.DotNet.DesktopRuntime.6

$directory = "C:\OpenTabletDriver"
if (!(Test-Path -Path $directory)) {
    New-Item -ItemType Directory -Path $directory
}

$url = "https://github.com/OpenTabletDriver/OpenTabletDriver/releases/latest/download/OpenTabletDriver.win-x64.zip"
$output = "$directory\OpenTabletDriver.win-x64.zip"
Invoke-WebRequest -Uri $url -OutFile $output

$destination = $directory
Expand-Archive -Path $output -DestinationPath $destination -Force
```

If you do want it to be system-wide, you can just run the `convert_to_portable.bat` file.

LRSFC-DanJ, 2024-05-24:
That's really helpful, thank you. Is there a command line switch for the "daemon" executable to install it as a Windows Service, or should we just use a service wrapper such as "Shawl" for example?