OpenTabletDriver/OpenTabletDriver #3324: Windows. CTH-680. The driver application crashes on startup

Sh1kami, 2024-05-21:
### Steps to Reproduce

System.ArgumentNullException: Value cannot be null. (Parameter 'source')
   at System.Linq.ThrowHelper.ThrowArgumentNullException(ExceptionArgument argument)
   at System.Linq.Enumerable.Any[TSource](IEnumerable`1 source, Func`2 predicate)
   at OpenTabletDriver.UX.Controls.TabletSwitcherPanel.TabletSwitcher.<HandleTabletsChanged>b__12_0(TabletReference tablet) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\TabletSwitcherPanel.cs:line 127
   at System.Linq.Enumerable.WhereEnumerableIterator`1.MoveNext()
   at OpenTabletDriver.UX.Controls.TabletSwitcherPanel.TabletSwitcher.HandleTabletsChanged(Object sender, IList`1 tablets) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\TabletSwitcherPanel.cs:line 130
   at OpenTabletDriver.UX.Controls.TabletSwitcherPanel.HandleTabletsChanged(Object sender, IEnumerable`1 tablets) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\TabletSwitcherPanel.cs:line 73
   at OpenTabletDriver.UX.RPC.DaemonRpcClient.<>c__DisplayClass13_2.<OnConnected>b__6() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\RPC\DaemonRpcClient.cs:line 32
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

### Expected Behavior

Starting the program

### Observed Behavior

Program crash

### Tablet Device

CTH-680

### Diagnostic Information

System.ArgumentNullException: Value cannot be null. (Parameter 'source')
   at System.Linq.ThrowHelper.ThrowArgumentNullException(ExceptionArgument argument)
   at System.Linq.Enumerable.Any[TSource](IEnumerable`1 source, Func`2 predicate)
   at OpenTabletDriver.UX.Controls.TabletSwitcherPanel.TabletSwitcher.<HandleTabletsChanged>b__12_0(TabletReference tablet) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\TabletSwitcherPanel.cs:line 127
   at System.Linq.Enumerable.WhereEnumerableIterator`1.MoveNext()
   at OpenTabletDriver.UX.Controls.TabletSwitcherPanel.TabletSwitcher.HandleTabletsChanged(Object sender, IList`1 tablets) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\TabletSwitcherPanel.cs:line 130
   at OpenTabletDriver.UX.Controls.TabletSwitcherPanel.HandleTabletsChanged(Object sender, IEnumerable`1 tablets) in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\Controls\TabletSwitcherPanel.cs:line 73
   at OpenTabletDriver.UX.RPC.DaemonRpcClient.<>c__DisplayClass13_2.<OnConnected>b__6() in D:\a\OpenTabletDriver\OpenTabletDriver\OpenTabletDriver.UX\RPC\DaemonRpcClient.cs:line 32
   at System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   at System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-07-30:
This probably has resolved itself by now, but if it hasn't please get the logs from the daemon (instead of just these ones from the UX).

jamesbt365, 2024-08-08:
Will close as stale, feel free to reopen if you still are encountering this.

closed 2024-08-08 by jamesbt365