OpenTabletDriver/OpenTabletDriver #3325: Support Request: TabletPC cant be Disabled and it interfears with the OpenTablet Driver

tsukYYnethack, 2024-05-24:
### What do you need help with?

I want to make OpenTablet's Driver override TabletPC or to disable TabletPC in some way. The bad thing is that on Windows11 you cant dissable the component, it doesnt even show up.

### What have you tried?

Restarting the computer, editing/deleting reg files and sys files (nothing that i deleted impacts the system only the TabletPC0, Searched for the TabletPC driver/svc/files/app/reg. Tried recording the events when i plug in my tablet after restart to see what and where is Tablet PC doing and the location of it. I have tried to Reinstall Windows and still nothing. And made OpenTablet's Driver run on Startup.

### Diagnostic Information

[diag.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/15433207/diag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

closed 2024-05-24 by tsukYYnethack

jamesbt365, 2024-05-24:
User error. The Wacom driver was installed.

tsukYYnethack, 2024-05-24:
Its the Wacom drivers. Im restarted :D 😆

tsukYYnethack, 2024-05-24:
I didnt know that wacom was doing it

tsukYYnethack, 2024-05-24:
ty