OpenTabletDriver/OpenTabletDriver #3330: Huion H1060P: Pressing multiple aux buttons simultaneously isn't handled properly

AnthonyMichaelTDM, 2024-05-31:
### Steps to Reproduce

Minimal preproducable example:

system: Arch linux

1. go to the Auxillary Settings tab and set one of the bindings to be (for example) Shift, and another to be (for example) R
2. apply
3. open a text editor or something that let's you capture inputs
4. press the two previously configured buttons at the same time (or press one while the other is being held) 

### Expected Behavior

I'd expect the simultaneous key presses to be handled properly, so in the above example I'd expect to see "R" inputted into the text editor.

### Observed Behavior

depending on which button was pressed first, either "r" is repeated or nothing is printed (because only the Shift button is being registered)

### Tablet Device

Huion H1060P

### Diagnostic Information

[diagnostics.json](https://github.com/user-attachments/files/15507513/diagnostics.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

AnthonyMichaelTDM, 2024-05-31:
I'm hoping this is actually an issue with OpenTabletDriver and not a hardware limitation of my tablet, but I'm unsure which is the case.

jamesbt365, 2024-05-31:
Tablet > tablet debugger, press the buttons on your tablet, does the data update properly allowing 2 buttons to be pressed at once?

AnthonyMichaelTDM, 2024-05-31:
It doesn't appear to

AnthonyMichaelTDM, 2024-05-31:
Does that mean it's a hardware limitation?

jamesbt365, 2024-05-31:
If the data doesn't indicate that mutiple buttons cannot be pressed at once then yes it would be a hardware limitation, the raw data is from the tablet itself.

closed 2024-05-31 by jamesbt365

AnthonyMichaelTDM, 2024-05-31:
unfortunate

AnthonyMichaelTDM, 2024-05-31:
is there a way to make a button "toggle"? like, if I have a button that is bound to "b", is there a way to make it so that if I press that button it will output "b" until I press it again?

jamesbt365, 2024-06-01:
No, but you could always make a plugin for it.

AnthonyMichaelTDM, 2024-06-01:
I'll look into that, do you have any pointers for where/how I should start?

jamesbt365, 2024-06-01:
There isn't any plugin documentation, but other plugins, especially ones that use bindings are a good place to look. I have a couple of them on my profile, namely BSoD and DualActionBindings, but there are others plugins that create bindings. You can find a button to any of the plugins source codes in the plugin manager inside OTD.