OpenTabletDriver/OpenTabletDriver #3341: Add support for Gaomon M7

Sorrow-Scarlet, 2024-06-17:
### Manufacturer

Gaomon

### Product Name

M7

### Device Width

258.4

### Device Height

161.5

### Pressure Levels

8191 in Gaomon official driver

### Lines per Inch (LPI)

5080

### Additional Information

It is said to have 16384 pressure levels when you open official site.  
I just found that i can only  get 8191 in its official driver like this :
[image]

### Diagnostic Information

[2024.6.17export.txt](https://github.com/user-attachments/files/15866320/2024.6.17export.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Sorrow-Scarlet, 2024-06-18:
P.S. Maybe Gaomon's new M7 is 16384 Pressure Lvl, but mine is the old.

Sorrow-Scarlet, 2024-06-18:
# P.S.  
I just ask their support just now, they told me they use another exe to update driver and firmware so that M7 could support 16384 Pressure Level. I think these might be useful for you

# Software
[M7_T207_20240401.zip](https://github.com/user-attachments/files/15884897/M7_T207_20240401.zip)

RainyLave, 2024-09-13:
Yes, all M7 devices are supporting 16384 pressure now!

mengh04, 2025-03-13:
> Yes, all M7 devices are supporting 16384 pressure now!

I saw that you have already provided support. Did you encounter any issues? I'm also using the M7, and I really need this support. Could you tell me how to use the configuration file you wrote? I compiled it using your branch, but it still can't recognize my M7.

closed 2026-01-09 by Kuuuube