OpenTabletDriver/OpenTabletDriver #3344: Support for Gaomon M10K

GregTheTurtle, 2024-06-19:
### Manufacturer

Gaomon

### Product Name

M10K

### Device Width

255.06

### Device Height

159.84

### Pressure Levels

8192 

### Lines per Inch (LPI)

5080

### Additional Information

.

### Diagnostic Information

[M10K.json](https://github.com/user-attachments/files/15896964/M10K.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-06-20:
This tablet is already supported by OpenTabletDriver, I can't tell from the information provided if this is a variant already supported by our driver, but as it stands you have the Gaomon driver installed which prevents OpenTabletDriver from working.

Uninstall the gaomon driver and if it still doesn't work go to `tablets > device string reader` in the driver and type the following values:
9580
110
201

then send the request and copy the output here.

jamesbt365, 2024-07-30:
I'm going to close this issue until you respond with a diagnostic with the gaomon driver uninstalled or if it works with the gaomon driver uninstalled.

closed 2024-07-30 by jamesbt365