OpenTabletDriver/OpenTabletDriver #3345: Support Request: CTL-470 detected but not working

DJshelfmushroom, 2024-06-22:
### What do you need help with?

OpenTabletDriver detects my tablet, and when I go into the debugger, when i put my pen down it changes a number, but nothing else changes when I move it.

### What have you tried?

i tried restarting and running tabletdrivercleanup.

### Diagnostic Information

[tabletdiag.json](https://github.com/user-attachments/files/15938297/tabletdiag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

DJshelfmushroom, 2024-06-22:
Good to note that the default wacom drivers (both current and recommended for device) also didn't work.

jamesbt365, 2024-06-22:
If you have them installed now, uninstall them, preferably by running TDCleanup after.

After this, go to tablet > tablet debugger, move your pen around. If you don't raw data changing and a blue dot, then the problem lies external to us.

DJshelfmushroom, 2024-06-22:
the default drivers? yeah I ran TDCleanup.

DJshelfmushroom, 2024-06-22:
i see data changing but no dot so...

closed 2024-06-22 by DJshelfmushroom