OpenTabletDriver/OpenTabletDriver #3351: Support Request: Acepen AP 1060 not detected

Asaltamontes, 2024-07-03:
### What do you need help with?

Acepen AP 1060 not detected.

### What have you tried?

Of course I used the tool: TabletDriverCleanup

Thanks for the help

### Diagnostic Information

[info.json](https://github.com/user-attachments/files/16089995/info.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-07-03:
device manager > view device by container > find your tablet on the list, open the container and screenshot

Asaltamontes, 2024-07-03:
> Administrador de dispositivos > Ver dispositivo por contenedor > Busque su tableta en la lista, abra el contenedor y haga una captura de pantalla

[image]

Asaltamontes, 2024-07-04:
I tried with my laptop to do a clean configuration from the beginning with the same result... tablet not connected...
Some help??
thanks in advance!!

jamesbt365, 2024-07-31:
I need a string dump to fix this.

Tablet > device string reader
21827
129

dump all, save that file and send it here.

Asaltamontes, 2024-08-03:
> I need a string dump to fix this.
> 
> Tablet > device string reader 21827 129
> 
> dump all, save that file and send it here.

Attached is the requested file. Thank you!

[acepen_1060.txt](https://github.com/user-attachments/files/16482848/acepen_1060.txt)

Asaltamontes, 2024-11-14:
Hi, any news on this topic? Thanks

gonX, 2026-10-03:
I assume this still isn't fixed, since the diagnostics looks like a variant from our currently supported config - this diag is ProductID `129` which conflicts which a bunch of other rebranded UC-Logic 1060N's, whereas our expected Acepen AP1060 ProductID is `130`.

However, it should be possible to make a draft configuration with the device strings provided above.