OpenTabletDriver/OpenTabletDriver #3353: Add support for Gaomon Tablet_PD1320

coelhoruno, 2024-07-05:
### Manufacturer

Gaomon

### Product Name

Gaomon Tablet_PD1320

### Device Width

293.8mm

### Device Height

165.2mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

[GAOMON PD1320 13.3-inch Affordable Drawing Display for Artists](https://gaomon.net/products/pd1320-pen-display)

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.4.0",
  "Build Date": "12/22/2023",
  "Operating System": {
    "Name": "Win32NT",
    "Version": "10.0.22631.0",
    "Attributes": {
      "ServicePack": ""
    }
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\{redacted}\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\{redacted}\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\{redacted}"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c232#2&1a87f3a8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "Unknown Product Name",
      "SerialNumber": "",
      "FriendlyName": "Unknown Product Name",
      "VendorID": 1133,
      "ProductID": 49714,
      "InputReportLength": 9,
      "OutputReportLength": 2,
      "FeatureReportLength": 0,
      "CanOpen": false,
      "DeviceAttributes": {
        "HID_REPORTS": "00:0001:0006"
      }
    },
    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006d&mi_01&col01#9&1e70c2bc&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": null,
      "ProductName": "Gaomon Tablet_PD1320",
      "SerialNumber": "",
[... 2147 lines cut ...]
    },
    {
      "Time": "2024-07-04T22:55:34.6965504-03:00",
      "Group": "DeviceIOException",
      "Message": "Failed to get info.",
      "StackTrace": "   at HidSharp.Platform.Windows.WinHidDevice.RequiresGetInfo(GetInfoFlags flags)\r\n   at HidSharp.Platform.Windows.WinHidDevice.GetManufacturer()\r\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_Manufacturer>b__13_0(HidDevice d) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Devices\\HidSharpBackend\\HidSharpEndpoint.cs:line 27\r\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Extensions.cs:line 12",
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-07-04T22:55:34.7032763-03:00",
      "Group": "DeviceIOException",
      "Message": "Failed to get info.",
      "StackTrace": "   at HidSharp.Platform.Windows.WinHidDevice.RequiresGetInfo(GetInfoFlags flags)\r\n   at HidSharp.Platform.Windows.WinHidDevice.GetProductName()\r\n   at HidSharp.HidDevice.GetFriendlyName()\r\n   at OpenTabletDriver.Devices.HidSharpBackend.HidSharpEndpoint.<>c.<get_FriendlyName>b__17_0(HidDevice d) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Devices\\HidSharpBackend\\HidSharpEndpoint.cs:line 29\r\n   at OpenTabletDriver.Extensions.TryGet[TSource,TValue](TSource source, Func`2 predicate, TValue& value) in D:\\a\\OpenTabletDriver\\OpenTabletDriver\\OpenTabletDriver\\Extensions.cs:line 12",
      "Level": 0,
      "Notification": false
    }
  ]
}
```

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-07-05:
Tablet > device string reader
9580
109
dump all, save that file and upload it here.

coelhoruno, 2024-07-05:
[string_dump.txt](https://github.com/user-attachments/files/16104236/string_dump.txt)

coelhoruno, 2024-07-05:
(real interesting seeing so much Huion stuff there. It really is all the same shit)