OpenTabletDriver/OpenTabletDriver #3354: XP-Pen Star G960S Not Detected

SunRizzze, 2024-07-05:
### Steps to Reproduce

[image]
[image]
[image]

### Expected Behavior

Should be detected by the OTD

### Observed Behavior

Just connected fresh tablet, and no detection

### Tablet Device

XP-Pen Start G960S

### Diagnostic Information

[g960s.json](https://github.com/user-attachments/files/16112455/g960s.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-07-05:
Have you tried rebooting?

SunRizzze, 2024-07-05:
> Have you tried rebooting?, Rebuild your initramfs then reboot.

still the same.

jamesbt365, 2024-07-05:
export diagnostic again.

SunRizzze, 2024-07-05:
[xppen960s.json](https://github.com/user-attachments/files/16112839/xppen960s.json)

SunRizzze, 2024-07-05:
> export diagnostic again.

here:
[xppen960s.json](https://github.com/user-attachments/files/16112840/xppen960s.json)

SunRizzze, 2024-07-05:
> export diagnostic again.

[image]

btw, official driver work fine

jamesbt365, 2024-07-05:
Its not working on OTD because you have the XP-Pen driver installed.

closed 2024-07-05 by jamesbt365

SunRizzze, 2024-07-05:
> Its not working on OTD because you have the XP-Pen driver installed.

im installed it after OTD, to check if it is detected

jamesbt365, 2024-07-05:
An interface is completely missing, the one OTD wants to use, OTD cannot use that interface because another driver is using it.

SunRizzze, 2024-07-05:
> An interface is completely missing, the one OTD wants to use, OTD cannot use that interface because another driver is using it.

uninstalled pentablet driver, still the same

jamesbt365, 2024-07-05:
The interface that OTD is trying to access is not available, its not even visible to the driver because its in use elsewhere.

This problem is external to OpenTabletDriver it simply does not have the permissions because its already in use. There is nothing I can do about that, there is no change in the driver I can make, there is no change to the configuration I can make, the interface that OTD needs to use is **not available**.

Once its uninstalled, rebuild your initramfs and reboot again, I don't know if it made any changes to the modules on your system.

SunRizzze, 2024-07-06:
> The interface that OTD is trying to access is not available, its not even visible to the driver because its in use elsewhere.
>
> This problem is external to OpenTabletDriver it simply does not have [...]

[image]
here my old g430 easily detected
(Official Pen Tablet installed, but disabled)

jamesbt365, 2024-07-06:
As you are insistent on it being the drivers fault, run the driver as root.

SunRizzze, 2024-07-07:
> As you are insistent on it being the drivers fault, run the driver as root.

Same happens on Windows, where i didnt had PenTabletDriver installed

OTD just doesn't detect my G960S

jamesbt365, 2024-07-07:
Export diagnostics on windows.

SunRizzze, 2024-07-07:
> Export diagnostics on windows.

[windows g960s.json](https://github.com/user-attachments/files/16120643/windows.g960s.json)

here, i want OTD to work cuz i hate drivers from XP-Pen

jamesbt365, 2024-07-07:
You have a new variant, doing this will not fix it on Linux until you fix the issue with the interface not being available but this will remedy it on Windows.

Download [this](https://github.com/user-attachments/files/16120651/Star.G960S.json) file, create a folder called `Configurations` next to OpenTabletDriver, then place that file inside the folder, finally replug your tablet.

**Then** go to tablet > tablet debugger:
- get the highest x and y coordinates you can by moving your pen to the bottom right of your tablet
- get the max pressure by pressing the pen nib in
- check if the pen buttons go to true when you press them and check the tablet buttons for the same.

SunRizzze, 2024-07-07:
[image]
[image]
[image]
[image]

now detected

SunRizzze, 2024-07-07:
> You have a new variant, doing this will not fix it on Linux until you fix the issue with the interface not being available but this will remedy it on Windows.
>
> Download [...]

[image]

jamesbt365, 2024-07-07:
None of your screenshots verify the last 2 bullet points.

SunRizzze, 2024-07-07:
> None of your screenshots verify the last 2 bullet points.

yes, they all work
[image]

[image]

jamesbt365, 2024-07-07:
You still missed a bullet point.

SunRizzze, 2024-07-07:
[image]

pressure

SunRizzze, 2024-07-07:
> You still missed a bullet point.

btw, what was the issue?

jamesbt365, 2024-07-07:
Your original issue? Like I said, interface isn't there because of permissions.

The *current* issue? Your tablet didn't match the identifiers in OTD.