OpenTabletDriver/OpenTabletDriver #3355: G960S failed detection

SunRizzze, 2024-07-06:
[image]
here my old g430 easily detected
(Official PenTabletDriver installed, but disabled)

i hope that you can fix g960s detection

_Originally posted by @SunRizzze in https://github.com/OpenTabletDriver/OpenTabletDriver/issues/3354#issuecomment-2211710490_

jamesbt365, 2024-07-06:
Like I told you, this has nothing to do with the driver, please keep everything in one issue.

closed 2024-07-06 by jamesbt365

SunRizzze, 2024-07-06:
> Like I told you, this has nothing to do with the driver, please keep everything in one issue.

so it means that my issue will be ignored and not resolved?