OpenTabletDriver/OpenTabletDriver #3357: Add support for

Gak-den, 2024-07-07:
### Manufacturer

Wacom

### Product Name

Intuos Pro (M)

### Device Width

338

### Device Height

219

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

I could not find my tablet model in the list of supported tablets. Whether or not I stay with Linux depends on this.
I would really appreciate it!
I tried looking for the .json file after exporting diagnostics, but I just can't find it wherever I save it... It's nothing really worth looking at, anyway. The best I could do was take a screenshot.

### Diagnostic Information

[image]

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-07-07:
You are using the Flatpak. The flatpak does not work straight of the box, unless official packages are not present on your distro it is highly recommended to use that instead.

If you must use the Flatpak see the manifest for setup instructions, these are mandatory.

Additionally, OTD has support fir this tablet, the support list uses the model numbers, not the friendly names else we would have multiple instances of intuos pro.

I'd also very much appreciate if you followed what the checkboxes said, instead of blindly following them. That is not a cinsise and meaningful title.

closed 2024-07-07 by jamesbt365

Gak-den, 2024-07-08:
Ah, sorry for the inconvenience! I thought "Intuos Pro" was important, as
it differentiates from tablets that have a screen like Cintiqs... I should
have been able to figure that out on my own.

Yes, it is indeed flatpak, now that you mention it. Guess I'll make a
mental note to verify that in my terminal from now on if I'm not sure. My
distro's official store doesn't always specify, and Ubuntu's Launchpad
doesn't list a PPA for an official package...
I didn't install the .deb file that OpenTablet provides because I wanted to
receive any future updates, but I suppose I could always check back to see
if there's a new version available for upgrade.

As I mentioned, I tried to get a hold of the diagnostics .json file, but
every time I tried looking for it wherever I saved it in my files, the
system simply wouldn't see it, as if it was unrecognizable, so I tried
installing a text package called Vim, and still I had no luck... I'll check
the manifest for that as well if I can find it, as the only available
version for my distro is flatpak. I'm new to Linux and still in the process
of learning things, so I thought there weren't extra steps to take for that
kind of package.
   Perhaps I should have written "Wacom Intuos Pro PTH660 support" for the
title...

   Thank you for the advice!

jamesbt365, 2024-07-08:
> As I mentioned, I tried to get a hold of the diagnostics .json file, but
every time I tried looking for it wherever I saved it in my files, the
system simply wouldn't see it

Flatpak's are containerised by default, everything is in `~/.var`.