OpenTabletDriver/OpenTabletDriver #3373: Add support for XPPen Deco 640

mczph, 2024-07-11:
### Manufacturer

XPPen

### Product Name

Deco 640

### Device Width

188.34mm

### Device Height

149.96mm

### Pressure Levels

16384

### Lines per Inch (LPI)

5080

### Diagnostic Information

[tablet.txt](https://github.com/user-attachments/files/16176450/tablet.txt)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-07-11:
You using the `Administrator` account makes supporting tablets challenging due to the lack of the ability to use the inbuilt debugging tools.

This is not supported by OTD and I will be unable to make you a configuration unless you can stop running OTD as admin, which is not possible on the account called Administrator.

mczph, 2024-07-11:
[tablet.txt](https://github.com/user-attachments/files/16177347/tablet.txt)
@jamesbt365 I have rerun the diagnostic with another account which is not an admin

DerAnsari, 2025-09-22:
hi 
is this completed? is the support for it added and if so how?

gonX, 2025-09-22:
> hi is this completed? is the support for it added and if so how?

You can either compile OpenTabletDriver from source, or wait for the next release. The milestone has just finished, so the release is planned to come out some time this week.

DerAnsari, 2025-09-22:
awesome !!

I just recent got this tablet and I really wanted to play Osu!lazer with it, thank you.

PR #3589 (2025-02-10): Add support for XP-Pen Deco 640 (IT640)
Fixes #3373 

Tested and working on Windows.
I did try to run OTD on my laptop with Fedora 41 but dotnet-6.0 isn't supported so I gave up. I can try with another distro if that's needed.

[windows-diagnostics.json](https://github.com/user-attachments/files/18102082/windows-diagnostics.json)
[tablet-data.txt](https://github.com/user-attachments/files/18102083/tablet-data.txt)
PR #3801 (2025-09-02): [Backport] Add support for XP-Pen Deco 640 (IT640)
Fixes #3373 for 0.6.x

Backport of #3589