OpenTabletDriver/OpenTabletDriver #3375: opentabletdriver.8 manpage is out of date

jamesbt365, 2024-07-12:
we have changed our packaging and webpage a lot and thus the manpage provides outdated information.

InfinityGhost, 2024-07-16:
Maybe this should be added an automated process

jamesbt365, 2024-08-25:
I'm not really sure how we would automate this, or what is the right way to structure manpages, if anybody wants to have a look at updating this for now and we can possible look at making the process smoother in the future that should suffice.

gonX, 2024-12-12:
My 2cents: delete the irrelevant stuff and redirect to wiki. Too many things are too likely to change with Avalonia.

gonX, 2024-12-15:
Fixed for 0.6.5, moving to 0.7.0 milestone instead since I think we still need to improve how this is done.