OpenTabletDriver/OpenTabletDriver #3378: Cursor jumps to random location upon release or click under Relative Mode

asrieldashie, 2024-07-16:
### Steps to Reproduce

1. Set to Relative Mode
2. Move the pen

https://github.com/user-attachments/assets/444b5b1e-f29a-4c1b-baee-3a62dff83f7e

### Expected Behavior

Does not jump to random locations upon release or click

### Observed Behavior

Jumps to random locations upon release or click

### Tablet Device

XP-Pen Star 03 V2 (Shows as UC-Logic 1060N in OTD)

### Diagnostic Information

[diagnostics.txt](https://github.com/user-attachments/files/16244532/diagnostics.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-07-16:
Try resetting your settings to default (file > reset to defaults and save), then set yourself into relative mode and again and see if it works. If it still doesn't, try reducing your reset time to 15ms. (make sure you save when you change your settings)

asrieldashie, 2024-07-16:
> Try resetting your settings to default (file > reset to defaults and save), then set yourself into relative mode and again and see if it works. If it still doesn't, try reducing your reset time to 15ms. (make sure you save when you change your settings)

Just reset and set the time to 15ms, but the issue still persist

sijesimochi, 2024-07-19:
It seems like it's conflicting with filters. I came across the same problem, and since I'm only using relative mode for general purpose don't really care about performance (latency, etc), I just turned all my filters off and it worked like a charm.

asrieldashie, 2024-07-19:
I have no filters installed nor even any plugin

jamesbt365, 2024-07-30:
> I have no filters installed nor even any plugin

Your original diagnostic shows at least 2 filters installed.

lotsofone, 2025-07-04:
I have the same problem with my UGEE S640 tablet. There is a change, maybe 1 out of 10 times, that the cursur jumps to somewhere nearby when I click a pen button. And it happens both with official driver and OpenTabletDriver. I guess it's not a driver problem. Maybe it caused by some EM interference between the buttons and the tablet?

gonX, 2025-08-27:
A bunch of assorted MacOS fixed was introduced with 0.6.5.1. 
Please retest with the latest version and get back to us. Also, please ensure that you do not have any filters activated.

Closing as stale until we hear a response.

closed 2025-08-27 by gonX