OpenTabletDriver/OpenTabletDriver #3380: Support for Wacom CTH-461 (same as CTH-460)

nclm, 2024-07-17:
### Manufacturer

Wacom

### Product Name

CTH-461

### Additional Information

As far as I know, the CTH-461 is exactly the same as the CTH-460 but in another case colour (and originally sold with some extra software). Since you have support for CTH-460, adding the CTH-461 could be just a matter of duplicating some config, or supporting both under the same code?

### Diagnostic Information

I don’t have the software installed yet.

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-07-17:
https://opentabletdriver.net/Tablets?search=CTH-461

closed 2024-07-17 by jamesbt365

nclm, 2024-07-17:
Ah somehow it wasn’t listed next to the CTH-460.
But this says that it does not support the buttons on the CTH-461. The the drivers does support the buttons of the CTH-460. Since I’m pretty certain both are the same hardware, could this issue be about adding button support to the CTH-461 since it is the same as the CTH-460?

jamesbt365, 2024-07-17:
#2207.

nclm, 2024-07-17:
Okay thanks, I don’t understand the technical bits there, so does it mean buttons are supported on the 461 like on the 460 and 661 and it’s just the website that isn’t to date? I’m not at home for the next few weeks to try out.

jamesbt365, 2024-07-17:
No you can't use them, the PR I linked removes the identifiers because *we don't have a suitable parser* for the aux buttons on that tablet.

nclm, 2024-07-17:
Oh okay thanks. It’s weird that the one for the CTH-460 doesn’t work for the CTH-461!

nilsherzig, 2024-07-22:
Excuse my ignorance, but is it a big undertaking to add support for the CTH-461 buttons? @jamesbt365 

If it's just a case of missing data, I would be happy to provide any logs / sniffing you might need :)

I really appreciate the plug and play nature of the pencil support, so far.