OpenTabletDriver/OpenTabletDriver #3381: XP-Pen Artist 12 Pro mapping display area incorrectly (multi monitor setup)

TheSHEEEP, 2024-07-18:
### Steps to Reproduce

1. Installed OTD according to instructions.
2. Connected tablet, it is detected correctly, both by KDE Plasma & OTD.
3. In the System Settings under Display Configuration, set the device to be a replica of my main screen (I don't want it as a third screen), like this:
[image]

4. Configure OTD like this:
[image]

### Expected Behavior

I'd expect the pen movement to adhere to the OTD area settings as shown above.

### Observed Behavior

Instead of the expected behavior, the pen position matches the mouse position only in the top left corner of the tablet.  
The further I move the pen away from that position, the more off the mouse cursor becomes (in the bottom-right direction). 

It is as if the distance to the top-left corner is applied twice.

I did install the official XP-Pen drivers first and tried to use those. The display mapping worked just fine there!  
Unfortunately, the keyboard mapping in the XP-Pen Linux driver for the side panel buttons is complete garbage, only 2/8 buttons work and the wheel not at all (which is alright, afaik that wheel works nowhere in Linux).
  
Of course, I removed the XP-Pen drivers & restarted before installing and using OTD.
I was hoping that I'd have more luck with OTD with the side buttons, but it looks like it fails already in the display mapping. 

### Tablet Device

XP-Pen Artist 12 Pro

### Diagnostic Information

[diags.json](https://github.com/user-attachments/files/16287318/diags.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-07-18:
tablet > tablet debugger, move your pen to the bottom right of the physical tablet, get the max x and y coordinates.

TheSHEEEP, 2024-07-18:
They are [51267, 28829]

jamesbt365, 2024-07-18:
This is an ongoing issue with some XP-Pen tablets that have no visible differences except in the report descriptor.

You can place this file in `~/.local/share/OpenTabletDriver/Configurations` but as it stands #3011 and possibly some additional stuff must be done first. See #990 for more details.
[Artist 12 Pro.json](https://github.com/user-attachments/files/16287411/Artist.12.Pro.json)

closed 2024-07-18 by jamesbt365

TheSHEEEP, 2024-07-18:
Awww, too bad. 
Guess it's back to somehow trying to get the official driver to have the side buttons work correctly.

Btw, your website lists the device as supported, with only "Tilt and wheel are not yet supported." as missing features.  
That definitely gave me a wrong impression as clearly more than that is missing atm.
Or maybe only in multi monitor case, not sure.

jamesbt365, 2024-07-18:
I just listed you a temporary solution, the file, in that location.

> That definitely gave me a wrong impression as clearly more than that is missing atm.

We can't predict what tablets may or may not release with different variations.

TheSHEEEP, 2024-07-19:
> I just listed you a temporary solution, the file, in that location.
> 
Ah, I got that wrong, I understood that the other points would have to be fixed in addition to that file.  
With your file in that location (and after a restart, but I guess restarting the daemon would've done the trick), the mapping seems to work fine, thanks!

> > That definitely gave me a wrong impression as clearly more than that is missing atm.
> 
> We can't predict what tablets may or may not release with different variations.
>
Maybe? I can't judge that.
But the point was that the website was giving the wrong impression. 
Put yourself in the mind of a new user just wanting to check if/how his device is supported by OTD - based on "Tilt and wheel are not yet supported." it seems pretty clear, but in fact another step was necessary that required manual support by a dev.

Maybe just add a * to the XP-Pen devices with an additional "* might require additional manual adjustments" or sth like that would suffice IMO.