OpenTabletDriver/OpenTabletDriver #3384: Linux + Wayland: Right Click, Middle Click Bound to Pen Buttons Not Working on HS610

Eoin-ONeill-Yokai, 2024-07-20:
### Steps to Reproduce

When using `Linux Artist Mode`, go to pen settings and bind the pen side buttons to `Mouse Button Right` and `Mouse Button Middle`. 

### Expected Behavior

I believe that using `evtest` on the virtual device (in my case, `/dev/input/event29`) should output some kind of mouse input for right or middle click. Additionally, I expect those buttons to work on my sway desktop after having launched the daemon.

### Observed Behavior

No inputs are fired from virtual device. Binding those buttons to other commands (such as pen button) results in working buttons but they're not longer bound to my desired button configuration for non-art programs.

### Tablet Device

Huion HS610

### Diagnostic Information

[opentablet-diagnostics.json](https://github.com/user-attachments/files/16322554/opentablet-diagnostics.json)

### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [ ] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-07-20:
Not a bug, mouse buttons don't exist on pens and thus can't be used, use artist mode bindings instead.

See our FAQ: https://opentabletdriver.net/Wiki/FAQ/Linux#artist-mode-bindings

Eoin-ONeill-Yokai, 2024-07-20:
Ah, I think I missed this. So is there a way to configure OTD to switch between artist mode and non artist mode?

It's a bit cumbersome to have to manually switch between the two modes.

jamesbt365, 2024-07-20:
File > save as preset
File > presets

Eoin-ONeill-Yokai, 2024-07-20:
Ah, Ok I see. So I guess with my system I can configure OTD to apply the preset by calling `otd applypreset <name>` Once I've saved to automate the system a bit more on my sway environment. Nice.

Thanks for the help @jamesbt365

Mrcubix, 2024-07-21:
You could also use one of your tablet's aux keys to switch to another profile using `Preset Binding`, available in Plugins > Plugin Manager.

closed 2025-12-06 by gonX