OpenTabletDriver/OpenTabletDriver #3388: opentabletdriver.daemon high cpu usage macos

aseotuh, 2024-07-23:
### Steps to Reproduce

launch opentabletdriver

### Expected Behavior

for it to not eat up my cpu

### Observed Behavior

uses large amounts of my cpu, often almost the entirety

### Tablet Device

wacom ctl 472

### Diagnostic Information

[Untitled2.json](https://github.com/user-attachments/files/16355115/Untitled2.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

aseotuh, 2024-07-23:
i want to add that i also had to quit opentabletdriver as it started using almost 300% CPU according to activity monitor, not sure this is 100% accurate, but my temps were up to ~110º C

jamesbt365, 2024-07-23:
Whats your settings on all your active filters?

aseotuh, 2024-07-23:
[image]

jamesbt365, 2024-07-23:
Try disabling it, saving and restarting OTD.

Do you still see the increased cpu usage?

aseotuh, 2024-07-23:
it goes down to ~0.1% cpu usage after I disable it

aseotuh, 2024-07-23:
i kinda need the antichatter cuz my aim is just shaky af without it tho so disabling it isn't really an option for me, is there a way I could use antichatter and not have it eat up my CPU?

jamesbt365, 2024-07-23:
Closing in favour of #3389.

closed 2024-07-23 by jamesbt365

aseotuh, 2024-07-23:
what

jamesbt365, 2024-07-23:
There is no native timer for MacOS in the latest version of the driver, but there is in the unstable branch.

However, its unstable and there are no filters for it. Either do not use filters, accept the CPU usage or if you know how to program feel free to backport the native timer.

aseotuh, 2024-07-23:
when i turn off the filter opentabletdriver itself starts using half my CPU

jamesbt365, 2024-07-23:
Filters are not cleaned up between saves of OTD, this is why I said to restart the driver when I told you to originally turn it off.