OpenTabletDriver/OpenTabletDriver #3393: Aiptek Hyperpen 120000U

jimbolaya, 2024-07-27:
### Manufacturer

Aiptek

### Product Name

Hyperpen 12000U

### Device Width

12"

### Device Height

9"

### Pressure Levels

512

### Lines per Inch (LPI)

3048

### Additional Information

There's an existing Aiptek driver that covers many of the Aiptek models. Unfortunately, there are no user configuration tools for this driver. 

### Diagnostic Information

No Tablet Detected

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-07-27:
@jimbolaya the diagnostic information it asks for specifies that you **must** go to help > export diagnostics, Dave the file and send it here. It is impossible for us to support tablets without this information.

jimbolaya, 2024-07-27:
I'm not sure how to add diagnostics since the GUI gets stuck attempting to connect to the daemon. When I start otd-daemon, it says "No tablets were detected." 

otd-gui sits saying "Connecting to OpenTabletDriver Daemon ..." 

No option to get diagnostics as far as I can tell. 

Makes sense since I don't believe this tablet isn't supported.

jamesbt365, 2024-07-27:
No, it doesnt make sense simply because it isnt supported, you have followed out installations instructions on which include a section on autostarting the driver component.

It cannot connect to something you aren't running.

jimbolaya, 2024-07-27:
Reading through the instructions more carefully. I ran 
`systemctl --user enable opentabletdriver.service --now`

As the user, not as root or sudo. I  tried as root earlier and it complained, which is why I tried running the daemon manually. It also may be good to mention that after running that particular command, a reboot is required, not just logging out and logging back in.

Here is a debug log from my latest attempt.
[otd-diagnostics.json](https://github.com/user-attachments/files/16400433/otd-diagnostics.json)

jamesbt365, 2024-07-27:
I don't see the tablet on the list of devices. Ensure it is plugged in and you have **no other drivers** running.

jimbolaya, 2024-07-27:
Blacklisted "aiptek" driver, rebooted.

BTW, here's the lsusb for the device:
`
Bus 002 Device 003: ID 08ca:0010 Aiptek International, Inc. Tablet`

New Diagnostics file:
[otd-diagnostics-2.json](https://github.com/user-attachments/files/16400662/otd-diagnostics-2.json)

jamesbt365, 2024-07-27:
Still cannot be seen, if you have installed any systemwide packages to make your tablet work, uninstall them or we will not be able to proceed.

jimbolaya, 2024-07-27:
There's nothing else.

gonX, 2026-10-04:
The HID devices list provided only contains non-tablet devices. If the tablet isn't HID-compliant that might be why - in that case, the tablet cannot be supported at the moment.

jimbolaya, 2026-10-08:
> If the tablet isn't HID-compliant that might be why

Ah, pretty sure it's not. It's a pretty old tablet.