OpenTabletDriver/OpenTabletDriver #3397: Xppen star g430s Incorrect Inputs

od1n69, 2024-07-30:
### Steps to Reproduce

Literally use the xppen star g430s 

### Expected Behavior

Normal area feeling, shouldn't feel like a new area on pc restart

### Observed Behavior

The input feels and is off, I tried other tablet drivers (devocub) and nothing seems to be wrong with the tablet itself, so this is an OTD problem, my area feels a whole lot different than what it should be
Side note everytime I replug in my tablet or restart pc, it feels like a completely different area

### Tablet Device

Xppen star g430s 

### Diagnostic Information

N/A

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

od1n69, 2024-07-30:
Windows 10 btw

jamesbt365, 2024-07-30:
Diagnostic information actually is applicable.

jamesbt365, 2024-08-08:
Will close because diagnostic information has not been supplied and thus its impossible to diagnose. Please upload the information from help > export diagnostics if you still need help.

closed 2024-08-08 by jamesbt365