OpenTabletDriver/OpenTabletDriver #3405: Support Request: OTD ignores second monitor while in Artist Mode

cloudskater, 2024-08-01:
### What do you need help with?

I am running Linux Mint Cinnamon.

**Steps to Reproduce**
1. Connect second monitor through HDMI port
2. Configure OTD for use with that second monitor exclusively in Absolute Mode
3. Switch to Artist Mode while choosing the same screen

**Expected Behavior**
OTD lets me use my tablet on the second monitor.

**Observed Behavior**
No matter if I switch to the second monitor before switching to Artist Mode or after, the mode refuses to let me draw on that second screen. If I switch back to Absolute it immediately starts working again.

I have no other tablet drivers installed, including the one that is stock for the tablet.
I followed the instructions to install OTD, and it runs fine outside of this issue as far as I can tell.

**Tablet Device**
Veikk A15

### What have you tried?

I have tried restarting the drivers and my computer, to no avail. I managed to get OTD to work on this screen in artist mode for one night the other day, and it hasn't done so before or since. I have no idea what I did to make it work.

### Diagnostic Information

[export.json](https://github.com/user-attachments/files/16462413/export.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

cloudskater, 2024-08-02:
I figured out a way **around** the issue, though didn't actually fix the problem itself so I'm unsure if I should mark this as resolved.

If you right-click the orange box under "Display" and de-select "Lock to usable area", the perimeter in which your tablet will draw can be moved outside the boundaries of your screens or, in this case, what the computer **thinks** are your screens.

I selected one of the two displays in the same right-click options menu (the choice made little difference as the displays are similar in resolution) and moved the orange box to the right of where it was formerly restrained, placing its left edge up against the usual boundary.
[image]
This only aligned the right edge of the A15 with the middle of my second screen, however, so I right-clicked the orange area, switched to the both screens option (X Screen 0) to widen the X axis, and did the same as before, like this:
[image]
This finally let me use the tablet on my second monitor alone, with the edges of the tablet lining up with the edges of the screen. It may be a workaround, but at least it does work.
You can do what I did and copy the Y axis value to paste after you have the X axis where you want it, just to make sure it all lines up the same vertically, but it doesn't make much difference.

Hope this helps someone else!

jamesbt365, 2024-08-08:
Likely some misconfiguration within your system or DE/WM, was discussed over discord and only happens in artist mode, likely will be fixed with an implemention of full artist mode passthrough)

closed 2024-08-08 by jamesbt365

sezanzeb, 2024-11-13:
possibly related: https://github.com/OpenTabletDriver/OpenTabletDriver/issues/3218