OpenTabletDriver/OpenTabletDriver #3407: Add support for Wacom 24HD DTK - 2400 Cintiq

Robking21, 2024-08-02:
### Manufacturer

Wacom

### Product Name

DTK - 2400

### Device Width

769.3 mm

### Device Height

463.74 mm

### Pressure Levels

2048

### Lines per Inch (LPI)

5080

### Additional Information

Screen size - 24.1 inch (518 X 324 mm)
Pen tilt levels +60 levels
Reading Height - 5 mm (0.2 inch) or more, center
[ipi-en.pdf](https://github.com/user-attachments/files/16472306/ipi-en.pdf)

### Diagnostic Information

[Wacom_DTK-2400.json](https://github.com/user-attachments/files/16472385/Wacom_DTK-2400.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

Robking21, 2024-08-02:
Thank you for your work and consideration of this driver!

muchitto, 2024-10-24:
This would be awesome! It is not supported anymore by Wacom so stuff gets buggy.

rsr-at-mindtwin, 2025-01-03:
I joined the discord specifically to see if anyone was working on this. I think it should be compatible with the DTK-2200.

rsr-at-mindtwin, 2025-01-05:
I have been looking through the code trying to figure out how to do this, but am not having much luck.

Attached please find three files: the first is the exported diagnostics report from OpenTabletDriver's UI, and two which are MacOS enumerations of the USB devices. The Wacom tablet attached to a direct USB connection.  

[2025-01-04 opentabletdriver diagnostics (cintiq24).txt](https://github.com/user-attachments/files/18308928/2025-01-04.opentabletdriver.diagnostics.cintiq24.txt)

[cintiq24hd system_profiler SPUSBDataType.txt](https://github.com/user-attachments/files/18308899/cintiq24hd.system_profiler.SPUSBDataType.txt)

[2025-01-04 diagnostics - ioreg.txt](https://github.com/user-attachments/files/18308921/2025-01-04.diagnostics.-.ioreg.txt)

What I observe is that Driver.cd Detect() never sees this device - in fact, of all the devices in the file above, the only two devices it even considers in the foreach() are the keyboard and the camera. This is in contrast to the json diagnostic (above) which clearly detects the device.  Some guidance would be appreciated.

gonX, 2026-10-04:
There's a tablet configuration guide at https://opentabletdriver.net/Wiki/Documentation/ConfigurationGuide if you're still willing to tackle this @rsr-at-mindtwin 

Also it sounds like you're using the `avalonia` or `master` branch of OpenTabletDriver. Please implement it using a release version (`0.6.x` branch).