OpenTabletDriver/OpenTabletDriver #3415: Doesn't detect Wacom PTH-460 at all

ashuramaruzxc, 2024-08-04:
### Steps to Reproduce

1. Use xanmod kernel
2. Install opentablet driver on NixOS:
```nix
hardware.opentabletdriver = {
    enable = true;
    daemon.enable = true;
};
services.udev = {
    packages = builtins.attrValues {
    inherit (pkgs.gnome) gnome-settings-daemon;
    inherit (pkgs.gnome2) GConf;
    inherit (pkgs) opentabletdriver yubikey-personalization /*libwacom*/;
};
  ```

### Expected Behavior

Getting detected

### Observed Behavior

3. Rebuild system and reboot just in case
4. DO `sudo rmmod wacom` just in case
5. Open otd gtk
6. See the error that tablet isn't detected

### Tablet Device

Wacom PTH-460

### Diagnostic Information

Can't insert here, insert here for some reason, so i'll upload it as a file
[dsadsa.json](https://github.com/user-attachments/files/16487876/dsadsa.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [ ] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-08-04:
Udev rules have not been applied correctly, ensure you have rebooted since applying this configuration.

ashuramaruzxc, 2024-08-04:
I rebooted after rebuilding configuration
Still ain't detecting

jamesbt365, 2024-08-04:
Export diagnostics again.

ashuramaruzxc, 2024-08-04:
```json
{
  "App Version": "OpenTabletDriver v0.6.4.0",
  "Build Date": "08/01/2024",
  "Operating System": {
    "Name": "NixOS",
    "Version": "24.05 (Uakari)",
    "Attributes": {
      "BUILD_ID": "24.05.20240801.0540572",
      "ID": "nixos",
      "PRETTY_NAME": "NixOS 24.05 (Uakari)",
      "SUPPORT_END": "2024-12-31",
      "VERSION_CODENAME": "uakari",
      "VERSION_ID": "24.05",
      "KERNEL_VERSION": "6.6.43-xanmod1"
    }
  },
  "Environment Variables": {
    "USER": "ashuramaru",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": null,
    "PWD": "/Users/marie",
    "PATH": "/nix/store/q0krp6wdhzk65f3m0q0rwbih617rpy7l-dotnet-runtime-6.0.32/bin:/Users/marie/.local/bin:/run/wrappers/bin:/Users/marie/.local/share/flatpak/exports/bin:/var/lib/flatpak/exports/bin:/Users/marie/.nix-profile/bin:/Users/marie/.local/state/nix/profile/bin:/Users/marie/.local/state/nix/profile/bin:/etc/profiles/per-user/ashuramaru/bin:/nix/var/nix/profiles/default/bin:/run/current-system/sw/bin"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:3c:00.1/usb3/3-1/3-1:1.2/0003:1038:1838.0003/hidraw/hidraw2",
      "Manufacturer": "SteelSeries",
      "ProductName": "SteelSeries Aerox 3 Wireless",
      "SerialNumber": "",
      "FriendlyName": "SteelSeries Aerox 3 Wireless",
      "VendorID": 4152,
      "ProductID": 6200,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "2",
        "HID_REPORT_DESCRIPTOR_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:3c:00.1/usb3/3-1/3-1:1.0/0003:1038:1838.0001/hidraw/hidraw0",
      "Manufacturer": "SteelSeries",
      "ProductName": "SteelSeries Aerox 3 Wireless",
      "SerialNumber": "",
      "FriendlyName": "SteelSeries Aerox 3 Wireless",
      "VendorID": 4152,
      "ProductID": 6200,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true,
      "DeviceAttributes": {
        "USB_INTERFACE_NUMBER": "0",
        "HID_REPORT_DESCRIPTOR_NON_RECONSTRUCTABLE": "true"
      }
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:42:00.3/usb9/9-4/9-4.3/9-4.3:1.1/0003:056A:03DC.001C/hidraw/hidraw16",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Wacom Intuos Pro S",
      "SerialNumber": "3IQ0111000459",
      "FriendlyName": "Wacom Intuos Pro S",
      "VendorID": 1386,
      "ProductID": 988,
[... 2259 lines cut ...]
      "Group": "Detect",
      "Message": "No tablets were detected.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2024-08-04T17:39:14.1015015+03:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2024-08-04T17:39:15.5848527+03:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

jamesbt365, 2024-08-04:
Has been fixed by #3130, you have 3 options:
- drop [specifically this file](https://raw.githubusercontent.com/jamesbt365/OpenTabletDriver/868ec98c6810d08b46422aeb52c5597a679edd9c/OpenTabletDriver.Configurations/Configurations/Wacom/PTH-460.json) in your `~/.local/share/OpenTabletDriver/Configurations` and add the respective udev rules to your nix config
- Patch the OTD package for your configuration
- Wait until the next OTD release.

closed 2024-08-04 by jamesbt365