OpenTabletDriver/OpenTabletDriver #3428: Support Request: OpenTabletDriver doesn't open, but tablet clearly works in Osu!Lazer (Linux)

Cobeta-beta, 2024-08-16:
### What do you need help with?

OpenTabletDriver application doesn't open "completely". It suck on "Connecting to OpenTabletDriver Daemon...", after which, it gives a timeout error. Tablet works more then perfect in osu!lazer. So, the daemon is clearly running. And by saying "works more then perfect", I mean it. Since osu!lazer gives options to resize tablet area, which is works.
DE : Hyprland
distribution : Arch Linux (all updated.)
Tablet : XP-Pen star G430s

### What have you tried?

Restarting computer, recompiling OpenTabletDriver, reconnecting tablet.  

### Diagnostic Information

```
Aug 16 18:30:15 ThinPad systemd[1121]: Started OpenTabletDriver Daemon.
Aug 16 18:30:16 ThinPad systemd[1121]: opentabletdriver.service: Main process exited, code=exited, status=1/FAILURE
Aug 16 18:30:16 ThinPad systemd[1121]: opentabletdriver.service: Failed with result 'exit-code'.
Aug 16 18:30:19 ThinPad systemd[1121]: opentabletdriver.service: Scheduled restart job, restart counter is at 7.
Aug 16 18:30:19 ThinPad systemd[1121]: Started OpenTabletDriver Daemon.
Aug 16 18:30:19 ThinPad systemd[1121]: opentabletdriver.service: Main process exited, code=exited, status=1/FAILURE
Aug 16 18:30:19 ThinPad systemd[1121]: opentabletdriver.service: Failed with result 'exit-code'.
Aug 16 18:30:20 ThinPad systemd[1121]: Stopped OpenTabletDriver Daemon.
```
It repeats endlessly.

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-08-16:
Lazer has its own inbuilt version of OTD which runs within lazer itself, which explains why it is working inside there.

The systemd service is not starting properly likely because `DISPLAY` or `WAYLAND_DISPLAY` is not present in the systemd environment, the easiest way around this would be to just put `otd-daemon` in the startup applications of your WM/DE instead.

closed 2024-08-16 by jamesbt365

Cobeta-beta, 2024-08-16:
Didn't worked.

jamesbt365, 2024-08-16:
Is the daemon running?

Cobeta-beta, 2024-08-17:
Yep.

jamesbt365, 2024-08-17:
As in when looking in a process explorer you see `OpenTabletDriver.Daemon` running as your user and when you you try and run `otd-daemon` in a terminal again it says its already running?

Cobeta-beta, 2024-08-17:
Oh. Weirdly enough, it doesn't... Alright, thanks for help! :>