OpenTabletDriver/OpenTabletDriver #3430: No tablets are detected Gaomon s620

anatadake, 2024-08-19:
### Steps to Reproduce

[image]
[image]

### Expected Behavior

I want to play osu but the tablet doesn’t see it

### Observed Behavior

Doesn't see my tablet even though I did everything according to the instructions

### Tablet Device

Gaomon S620

### Diagnostic Information

[1.json](https://github.com/user-attachments/files/16661461/1.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-08-19:
Not a bug, you have the gaomon driver installed and from OTDs perspective your tablet isnt even plugged in

This could be because:
- You are using a USB Hub
- You are using a cable that doesnt support data transfer
- The gaomon driver is messing with the device drivers hiding it from OTD.

anatadake, 2024-08-19:
And what do I need to do

jamesbt365, 2024-08-19:
Uninstall the gaomon driver and ensure your tablet is connected properly, you can check its connected properly in device manager by going to `view > device by container`

jamesbt365, 2024-08-19:
By using the uninstaller in apps and features.

anatadake, 2024-08-19:
So, in theory, I have deleted everything related to Gaomon, what should I do next?

jamesbt365, 2024-08-19:
And how did you "delete" it, delete and uninstall mean different things.

anatadake, 2024-08-19:
I deleted it using unis

jamesbt365, 2024-08-19:
No, you *uninstalled* it using the uninstaller, deleting would be going into file explorer and manually deleting files, they are different terms and you don't delete installed applications, let alone drivers.

Is your device in device manager when you do the steps I previously provided?

anatadake, 2024-08-19:
No, it is not located

jamesbt365, 2024-08-19:
Then the tablet is not plugged in.

closed 2024-08-19 by jamesbt365

anatadake, 2024-08-19:
Should it be connected while deleting everything?

jamesbt365, 2024-08-19:
Again, you are using the wrong terminology, it is absolutely imperative that you are speaking clearly, you are supposed to *uninstall*, not *delete*.

It shouldn't matter, but it may be impossible for uninstallers to undo changes to device drivers when the device is not plugged in, regardless **the tablet must be plugged in for OpenTabletDriver** to use it anyway, so the end result is the same, the device **must be plugged in**.

anatadake, 2024-08-19:
I just don’t understand English well, that’s why I communicate through a translator
So I connected the tablet, what should I do next, I deleted gaomon everywhere it was located

jamesbt365, 2024-08-19:
If your device is **not present in device manager** when you go to **view > device by container** it is not plugged in, and therefore this issue has absolutely nothing to do with OTD.

I have already given you some possible remedies in this chat.

anatadake, 2024-08-19:
It's connected there

jamesbt365, 2024-08-19:
Screenshot.

anatadake, 2024-08-19:
[image]

jamesbt365, 2024-08-19:
If running this program and removing everything it asks for doesn't help, then there isn't anything else I can do here because of the language barrier, this problem is with your setup and doesn't have anything to do with OpenTabletDriver, at all.

https://opentabletdriver.net/Wiki/FAQ/Windows#driver-cleanup

anatadake, 2024-08-19:
I did everything you suggested but nothing helped