OpenTabletDriver/OpenTabletDriver #3435: Add support for Fansjoy S2 tablet

sekatol, 2024-08-22:
### Manufacturer

Fansjoy

### Product Name

FJ-S2

### Device Width

115mm

### Device Height

72mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Without any driver, this tablet sends wrong cursor position to system.
Like this:
```
+------+                   +-------------+
|      |                   |             |
|      | --map-to-screen-> |             |
|      |                   +-------------+
+------+                       screen
 tablet
```
But it should be like this: (rotate 270 degree clockwise)
```
+----------+                   +-------------+
|          | --map-to-screen-> |             |
|          |                   |             |
+----------+                   +-------------+
   tablet                          screen
```

### Diagnostic Information

[diag.json](https://github.com/user-attachments/files/16703013/diag.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

InfinityGhost, 2024-08-23:
Can you provide some sort of proof this thing exists? A pretty thorough search doesn't show anything at all.

sekatol, 2024-08-23:
Sorry for the inconvenience. This tablet is from a small corporation in China, and I bought it from Taobao, an online shopping platform in China. I found the [official website](https://www.fans-joy.com/s) of Fansjoy through Bing Chinese. (Sorry for no translations.) The official website has an entry for this model, "S2数位板" for S2 tablet. The button below it is "了解更多" means "Learn more", but it is just a link to Taobao site, 😂. I copied information on that website:
- Pressure Level: 8192
- Resolution: 5080 LPI
- Read Frequency: 266 points/second
- Max Tilt Angle: 60 degree
- Response Height: 1 cm
- 4 customizable buttons
- A LED light
- Additional information: Press the first and fourth button simultaneously for 6 seconds to toggle orientation. 
- There are 2 buttons on the digital pen. The upper one is for the mouse middle button and the lower one is for the eraser.
- No need to charge the pen.

I don't know if there is a need to build a driver for this tablet, because there are really a few people using this model of tablet, so working on this tablet is time wasting.
Or, if possible, I will try to learn from the recent pull requests and contribute codes for this tablet, giving any guidances are the best.

jamesbt365, 2024-08-25:
Move [this](https://github.com/user-attachments/files/16738736/70-opentabletdriver.rules.txt) to `/etc/udev/rules.d/70-opentabletdriver.rules` exactly as that filename is (meaning you remove the .txt from the file i am uploading, its only present because GitHub)

Then either run the following commands or reboot before exporting diagnostics again.
`sudo udevadm control --reload-rules && sudo udevadm trigger`

sekatol, 2024-08-26:
I moved the file into `/etc/udev/rules.d/`:
```bash
$ ls /etc/udev/rules.d/
70-opentabletdriver.rules  99-hwthrvop-permissions.rules
```
After rebooting, the new diagnostics is [here](https://github.com/user-attachments/files/16745185/diag2.json).