OpenTabletDriver/OpenTabletDriver #3449: Add support for

williamyooda, 2024-09-01:
### Manufacturer

xp pen

### Product Name

star 01

### Device Width

204

### Device Height

127

### Pressure Levels

2048

### Lines per Inch (LPI)

5080

### Additional Information

there a driver for it but too old for my win 11 os installed but wacky 

### Diagnostic Information

[xpen star 01.json](https://github.com/user-attachments/files/16829262/xpen.star.01.json)

### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

jamesbt365, 2024-09-01:
>  I have written a concise and meaningful title.

No you didn't.

williamyooda, 2024-09-01:
sorry its my first time use git hub .

williamyooda, 2024-09-22:
hey , i am very curious how the process to make a driver for pen tablet . can you tell me what the process to make a open driver for pentablet .

williamyooda, 2024-09-22:
sorry miss click