OpenTabletDriver/OpenTabletDriver #3452: Driver maps incorrect coordinates with Dual monitors in Hyprland

Artrix9095, 2024-09-05:
### Steps to Reproduce

https://github.com/user-attachments/assets/ca0ef599-65d9-438d-ad1c-34fb89b2db50

### Expected Behavior

That the driver maps my whole screen to my tablet (using the same exact settings on windows and it works fine)

### Observed Behavior

As shown in the video, when reaching the left edge of the tablet (where it should be the left edge of my screen) it stops roughly 70% through the monitor. My second monitor also happens to be to the left. I also noticed that on the right side of my tablet it hits the edge of my screen when my pen is still about 5mm away from the edge of my tablet. All this is happening when I'm on full area, and using a perfectly mapped output to my main monitor.

### Tablet Device

Wacom CTL-472

### Diagnostic Information

Here is my `hyprctl monitors`
```
Monitor DP-2 (ID 0):
	2560x1440@240.00101 at 1080x390
	description: Acer Technologies XV272U W2 B414027E94220
	make: Acer Technologies
	model: XV272U W2
	serial: B414027E94220
	active workspace: 1 (1)
	special workspace: 0 ()
	reserved: 0 45 0 0
	scale: 1.00
	transform: 0
	focused: yes
	dpmsStatus: 1
	vrr: 1
	activelyTearing: false
	disabled: false
	currentFormat: XRGB8888
	availableModes: 2560x1440@240.00Hz 2560x1440@59.95Hz 2560x1440@144.00Hz 2560x1440@120.00Hz 1920x1200@240.00Hz 1920x1080@239.96Hz 1920x1080@120.00Hz 1920x1080@119.88Hz 1920x1080@100.00Hz 1920x1080@60.00Hz 1920x1080@60.00Hz 1920x1080@59.94Hz 1920x1080@50.00Hz 1920x1080@30.00Hz 1920x1080@29.97Hz 1920x1080@25.00Hz 1920x1080@24.00Hz 1920x1080@23.98Hz 1600x1200@240.00Hz 1680x1050@59.95Hz 1280x1024@75.03Hz 1280x1024@60.02Hz 1440x900@59.89Hz 1280x960@60.00Hz 1280x800@240.00Hz 1152x864@75.00Hz 1280x720@60.00Hz 1280x720@59.94Hz 1280x720@50.00Hz 1024x768@119.93Hz 1024x768@75.03Hz 1024x768@70.07Hz 1024x768@60.00Hz 1440x480@60.00Hz 1440x480@60.00Hz 1440x480@59.94Hz 1440x480@59.94Hz 832x624@74.55Hz 800x600@75.00Hz 800x600@72.19Hz 800x600@60.32Hz 800x600@56.25Hz 720x576@50.00Hz 720x576@50.00Hz 720x480@60.00Hz 720x480@60.00Hz 720x480@59.94Hz 720x480@59.94Hz 640x480@75.00Hz 640x480@72.81Hz 640x480@66.67Hz 640x480@60.00Hz 640x480@59.94Hz 640x480@59.94Hz 720x400@70.08Hz

Monitor DP-3 (ID 1):
	1920x1080@164.91701 at 0x0
	description: ASUSTek COMPUTER INC VG248 LALMQS129671
	make: ASUSTek COMPUTER INC
	model: VG248
	serial: LALMQS129671
	active workspace: 2 (2)
	special workspace: 0 ()
	reserved: 0 0 0 0
	scale: 1.00
	transform: 1
	focused: no
	dpmsStatus: 1
	vrr: 1
	activelyTearing: false
	disabled: false
	currentFormat: XRGB8888
	availableModes: 1920x1080@119.98Hz 1920x1080@164.92Hz 1920x1080@144.00Hz 1920x1080@99.93Hz 1920x1080@84.91Hz 1920x1080@60.00Hz 1920x1080@60.00Hz 1920x1080@59.94Hz 1920x1080@50.00Hz 1680x1050@59.95Hz 1280x1024@75.03Hz 1280x1024@60.02Hz 1440x900@59.89Hz 1280x960@60.00Hz 1280x800@59.81Hz 1152x864@75.00Hz 1280x720@60.00Hz 1280x720@60.00Hz 1280x720@59.94Hz 1280x720@50.00Hz 1024x768@75.03Hz 1024x768@70.07Hz 1024x768@60.00Hz 832x624@74.55Hz 800x600@75.00Hz 800x600@72.19Hz 800x600@60.32Hz 800x600@56.25Hz 720x576@50.00Hz 720x576@50.00Hz 720x480@60.00Hz 720x480@60.00Hz 720x480@59.94Hz 720x480@59.94Hz 640x480@75.00Hz 640x480@66.67Hz 640x480@60.00Hz 640x480@59.94Hz 640x480@59.94Hz 720x400@70.08Hz
```

Diagnostic output:
[Diagnostic.json](https://github.com/user-attachments/files/16901246/otd.txt)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-09-05:
hyprwm/Hyprland#6023.

closed 2024-09-05 by jamesbt365

claymorwan, 2025-09-14:
im issues like this as well