OpenTabletDriver/OpenTabletDriver #3461: Device Disconnected right after connecting Tablet

mgs0201, 2024-09-11:
### Steps to Reproduce

1. Connect tablet via USB 2.0 (this case is XPPen Deco Fun XS)
2. Try to reconnect a few times, it won't work

### Expected Behavior

OTD should show the device connected and manipulate parameters

### Observed Behavior

OTD logs driver is enabled but instantly disconnect the device. Note that the tablet keeps working as expected & it's recognized in **lsusb**, but it isn't in games like OSU!
[image]
.
.
.
[otd.log](https://github.com/user-attachments/files/16965854/otd.log)

### Tablet Device

XPPen Deco Fun XS

### Diagnostic Information

[otddiag.json](https://github.com/user-attachments/files/16965900/otddiag.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

jamesbt365, 2024-09-11:
For OpenTabletDriver to work correctly on Linux you must:
- You must not have any other tablet drivers installed
- You must have installed OpenTabletDriver from any of the official packages (for the udev and kernel module blacklists)
- You must have a working udev setup.

mgs0201, 2024-09-11:
`You must not have any other tablet drivers installed`

Located the issue. XPPen driver app interferes with OTD. Closing or uninstalling the app solves the issue. Thanks

closed 2024-09-11 by mgs0201