OpenTabletDriver/OpenTabletDriver #3464: Issue with Huion Q630M Device Name in Configuration

Ac-Ada-Lovelace, 2024-09-15:
### Steps to Reproduce

Connect Huion Q630M tablet to the system.
Run OpenTabletDriver's diagnostic function.
Check the device name fields in the output compared to the configuration file.

### Expected Behavior

The tablet should be recognized without issues.

### Observed Behavior

The tablet is not properly recognized due to a possible mismatch between the ProductName and the Name field in the configuration file.

### Tablet Device

Huion Q630M

### Diagnostic Information

I am experiencing an issue with my Huion Q630M tablet while using OpenTabletDriver. Although this device is listed as supported, I have encountered problems when trying to configure it.

According to the diagnostic output, the device fields are as follows:

```
"ProductName": "Huion Tablet_Q630M",
"SerialNumber": "20200922",
"FriendlyName": "Huion Tablet_Q630M"
```
However, in the source code, the configuration file for the Huion Q630M lists the device name as:
```
"Name": "Huion Q630M"
```
It seems that there is a mismatch between the actual ProductName field ("Huion Tablet_Q630M") and the name in the configuration file ("Huion Q630M"). This might be the reason the driver is not properly recognizing the tablet.
[3.json](https://github.com/user-attachments/files/17004769/3.json)

### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.
- [X] I am on the latest version of OpenTabletDriver.

Ac-Ada-Lovelace, 2024-09-15:
I have already attempted to manually change the recognition information in the configuration file and recompile the driver, but unfortunately, it did not resolve the issue.
And then I try to run the 'build.ps1' to build. But the binary executes with exceptions

Ac-Ada-Lovelace, 2024-09-15:
this is the log file

[2024-09-15 11_12_54Z.log.json](https://github.com/user-attachments/files/17005532/2024-09-15.11_12_54Z.log.json)

jamesbt365, 2024-09-15:
The device name has absolutely nothing to do with if a tablet detects or not.

Tablet > device string reader
9580
96

dump all, save that file and send that here.

Ac-Ada-Lovelace, 2024-09-15:
[q630m.json](https://github.com/user-attachments/files/17005542/q630m.json)
@jamesbt365 Is there anything else needed?

jamesbt365, 2024-09-15:
Yes, I need the file that is generated from the string dump, I do not need you to upload another diagnostic.

Ac-Ada-Lovelace, 2024-09-15:
[dump.txt](https://github.com/user-attachments/files/17005560/dump.txt)
the dump here

jamesbt365, 2024-09-15:
Just to let you know I get your messages regardless of if you mention me here, so please stop doing that, its just obnoxious.

You will get a working configuration when I have the time to do that.

Ac-Ada-Lovelace, 2024-09-15:
Thank you for your response, and I apologize for tagging you multiple times. I didn’t realize it could be overwhelming. I appreciate your work and understand that it takes time to address these issues. Thanks again for your efforts!

I managed to resolve the issue by modifying the configuration file. Specifically, I changed the following line:

json
复制代码
"201": "HUION_T216_\\d{6}$"
to:

json
复制代码
"201": "HUION_T22b_\\d{6}$"
This change allowed my Huion Q630M tablet to be correctly recognized by OpenTabletDriver. However, since I am not familiar with the entire source code of the project, I am not sure if removing the original line might affect compatibility for other users.

Would it be possible for you to review this change and initiate a pull request on my behalf? I would appreciate your expertise in ensuring that this change doesn't introduce any unintended issues.

Thank you for your time and support!
[Q630M.json](https://github.com/user-attachments/files/17006865/Q630M.json)

PR #3720 (2025-01-24): Add support for variant of Huion Q630M
I don't have verification, but as the user has not complained or done anything else on that post, its safe to say that it works fine. And if it doesn't work properly then we can always revert if someone complains.

Fixes #3464, string dump and diag is in that issue.